wasm-micro-runtime/wasm-micro-runtime #2316: an alternate "indirect" way to find "stack_sizes" is necessary for xip

yamt, 2023-06-26:


yamt, 2023-06-29:
the CI failure looks unrelated
```
Err:1 http://azure.archive.ubuntu.com/ubuntu jammy/universe amd64 kconfig-frontends-nox amd64 4.11.0.1+dfsg-6
  Could not connect to azure.archive.ubuntu.com:80 (52.154.174.208), connection timed out [IP: 52.154.174.208 80]
Err:2 http://azure.archive.ubuntu.com/ubuntu jammy/universe amd64 genromfs amd64 0.5.2-4
  Unable to connect to azure.archive.ubuntu.com:http: [IP: 52.154.174.208 80]
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/universe/k/kconfig-frontends/kconfig-frontends-nox_4.11.0.1%2bdfsg-6_amd64.deb  Could not connect to azure.archive.ubuntu.com:80 (52.154.174.208), connection timed out [IP: 52.154.174.208 80]
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/universe/g/genromfs/genromfs_0.5.2-4_amd64.deb  Unable to connect to azure.archive.ubuntu.com:http: [IP: 52.154.174.208 80]
```

yamt, 2023-07-03:
"spec test on nuttx" seems still failing. @wenyongh can you reopen this?

yamt, 2023-07-03:
```
============> run br_table failed with a non-zero return code 101

==================== LAST LOG of br_table ====================

  expected: 'SystemExit(1)'
  got: 'iwasm --heap-size=0 -v=0 --repl /tmp/tmprdma9ep7.aot
AOT module load failed: cannot apply relocation to text section for aot file generated with "--enable-indirect-mode" flag
```

br_table test case has a large br_table, which happens to be translated into relocations like:
```
000003dc R_RISCV_HI20             .Lswitch.table.aot_func_internal#15
000003e0 R_RISCV_LO12_I           .Lswitch.table.aot_func_internal#15
```
i'm not sure why it was not a problem in the previous code.

i'm not sure how to fix either.
maybe we can disable switch lowering as a workaround?

any idea?

wenyongh, 2023-07-03:
Maybe it is due to some passes added for XIP mode don't work again with LLVM-15.0, as issue #2328 reported by @no1wudi, I think we can wait until the PR is merged.

no1wudi, 2023-07-03:
> 
> i'm not sure how to fix either. maybe we can disable switch lowering as a workaround?
> 

We should enable switch lowering to fix this problem, as @wenyongh mentioned, `LowerSwitch` pass is not actually working now with LLVM 15.

I'll fix it in #2334.

yamt, 2023-07-04:
> > i'm not sure how to fix either. maybe we can disable switch lowering as a workaround?
> 
> We should enable switch lowering to fix this problem, as @wenyongh mentioned, `LowerSwitch` pass is not actually working now with LLVM 15.
> 
> I'll fix it in #2334.

i meant something like https://github.com/bytecodealliance/wasm-micro-runtime/pull/2339
i understand your approach can fix it too.
i'm not sure which is a better approach.

no1wudi, 2023-07-04:
Maybe we should try #2339, LowerSwitch is alwasy convert switch to if ... else group, but not all switch will create jump table, this attribute seems keep small switch, which has less impact for performance and code size.

wenyongh, 2023-07-04:
So let's merge #2339 first and then #2334?

no1wudi, 2023-07-04:
> So let's merge #2339 first and then #2334?

Yes, I guess it's OK.

yamt, 2023-07-06:
the test is now green. https://github.com/bytecodealliance/wasm-micro-runtime/actions/workflows/spec_test_on_nuttx.yml

closed 2023-07-06 by yamt