wasm-micro-runtime/wasm-micro-runtime #2319: Support threads(cluster) suspension/resume

wenyongh, 2023-06-27:
## Motivation
Some applications/scenarios require that the threads in a cluster run into the suspension state, or is to suspend all threads except the thread which is requesting the suspension. Normally it is named the `Stop The World` event, meaning that all application threads are stopped until the operation completes. These scenarios may include:
- A thread is doing Garbage Collection and requests other threads to stop
- A thread enters into Breakpoint and requests other others to stop
- A standalone thread (not in the cluster) is to dump the snapshot (checkpoint) and request all threads in the cluster to stop

## Current Status
Currently WAMR implements several APIs:
```C
void wasm_cluster_suspend_all(WASMCluster *cluster);
void wasm_cluster_suspend_all_except_self(WASMCluster *cluster, WASMExecEnv *exec_env);
void wasm_cluster_suspend_thread(WASMExecEnv *exec_env);
void wasm_cluster_resume_thread(WASMExecEnv *exec_env);
void wasm_cluster_resume_all(WASMCluster *cluster);
```
but the implementation is simple and cannot meet the requirements: thread A can send the suspension request to thread B, but it doesn't wait until thread B actually stops, and thread B doesn't notify thread A when it stops.

## Approach
Introduce new locks/conds in cluster and new fields in exec_env:
```
WASMCluster.thread_state_lock: 
    The lock for thread_safe_cond, thread_resume_cond and the below fields of exec_env:
         suspend_flags, suspend_count, current_status, thread_ret_value

WASMCluster.thread_safe_cond:
    Condition variable for notifying threads that are waiting for threads to go to a safe state.
    All threads waiting for this condition share this one condition variable, so when waken up,
    they must check whether the waited threads are really in a safe state. It's protected by
    thread_state_lock since it involves operations of thread states.

WASMCluster.thread_resume_cond:
    Condition variable for notifying threads that are waiting for resumption that their suspend
    counts have changed and they may be able to resume (i.e. change to RUNNING state).
    All suspended threads share this one condition variable, so when waken up, they must
    recheck their suspend counts to check whether they are really able to resume. It's protected
    by thread_state_lock since it involves operations of suspend counts.

WASMExecEnv.wait_lock:
    The lock for protecting wait_cond, wait_count and thread_is_detached of this exec_env.

WASMExecEnv.wait_cond:
    Conditional variable for this thread to wait.
```
When a thread requests to suspend other threads, it firstly increases the suspend_count of other threads, and then try to acquire the lock for the exec_env_list, and then checks whether there is other thread suspending it, if yes, it must release the lock, suspend himself and wait until other thread wakeup it. And only when there isn't thread suspending it, can it lock the exec_enc_list, wait until all other threads are suspended, and then continue to perform the operation, e.g. Garbage Collection.

### Thread status
A thread can run into some status:
- Running: WASM_THREAD_RUNNING
- Waiting: WASM_THREAD_VMWAIT, thread is waiting for resource
- Suspended: WASM_THREAD_SUSPENDED, thread is suspended
- Zombie: WASM_THREAD_ZOMBIE, thread is detached or thread exited
And more status to support source debugging (TODO).

yamt, 2023-07-06:
do you have any plan how to "suspend" blocking threads?

wenyongh, 2023-07-06:
Not sure whether this meets your requirements: the thread can call the related API e.g. `wasm_runtime_enter_safe_state()` to enter into safe state before blocking, so other thread will ignore it when suspending other threads. And the former can call `wasm_runtime_exit_safe_state()` to go back into running state.

The feature is WIP, refer to: https://github.com/bytecodealliance/wasm-micro-runtime/pull/2320

yamt, 2023-07-06:
> Not sure whether this meets your requirements: the thread can call the related API e.g. `wasm_runtime_enter_safe_state()` to enter into safe state before blocking, so other thread will ignore it [...]

ah, i see.
i thought those functions were just unused leftover.
for some use cases, the approach should be ok.

btw, it has a wrong name. (`jeff_runtime_exit_safe_state`)
copy-and-paste mistake?
it reminded me that i have never understood what "jeff" in app manager is.

wenyongh, 2023-07-06:
Yes, it is a typo, part of the code were ported from an internal project.

yamt, 2023-07-06:
> Yes, it is a typo, part of the code were ported from an internal project.

does the internal project have a similar mechanism?
does it work well there?

wenyongh, 2023-07-06:
Yes, is has a similar mechanism and works well.