wasm-micro-runtime/wasm-micro-runtime #2330: failed to link import function (env, _ZN10vsomeip_v37runtime3getEv)

Anilkumar1818, 2023-06-30:
# Copyright (C) 2022 Intel Corporation.  All rights reserved.
# SPDX-License-Identifier: Apache-2.0 WITH LLVM-exception

cmake_minimum_required(VERSION 3.0)
project(wasm-app)
#find_package(vsomeip3)
set (WAMR_ROOT_DIR ${CMAKE_CURRENT_SOURCE_DIR}/../../..)

set (WAMR_BUILD_PLATFORM "linux")

# Reset default linker flags
# set (CMAKE_SHARED_LIBRARY_LINK_C_FLAGS "")
# set (CMAKE_SHARED_LIBRARY_LINK_CXX_FLAGS "")

# set (CMAKE_CXX_STANDARD 14)

if (APPLE)
    set (HAVE_FLAG_SEARCH_PATHS_FIRST 0)
    set (CMAKE_C_LINK_FLAGS "")
    set (CMAKE_CXX_LINK_FLAGS "")
endif ()
  include_directories(/home/anil/Downloads/vsomeip-3.3.0/build/out/include)
#   link_directories(/home/anil/Downloads/vsomeip-3.3.0/build/out/lib)
set (CMAKE_SYSTEM_PROCESSOR wasm32)

if (NOT DEFINED WASI_SDK_DIR)
    set (WASI_SDK_DIR               "/opt/wasi-sdk")
endif ()

#set (CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -Wextra -Wformat -Wformat-security -Wno-unused")

set (CMAKE_CXX_COMPILER_TARGET        "wasm32-wasi")
#set (CMAKE_CXX_COMPILER               "${WASI_SDK_DIR}/bin/clang++")
set (CMAKE_CXX_COMPILER "/opt/wasi-sdk/bin/clang++")

set (CMAKE_CXX_FLAGS                  "${CMAKE_CXX_FLAGS} -O3 -Wl,--allow-undefined")

add_executable(anil main_v.cpp)
target_link_libraries(anil)

#include <iomanip>
#include <iostream>

#include <vsomeip/vsomeip.hpp>
#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678

std::shared_ptr< vsomeip::application > app;

void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available) {
    std::cout << "Service ["
            << std::setw(4) << std::setfill('0') << std::hex << _service << "." << _instance
            << "] is " << (_is_available ? "available." : "NOT available.")  << std::endl;
    std::cout<<"hello";
}

int main() {
    std::cout << "vsomeip appp \n";
    app = vsomeip::runtime::get()->create_application("Hello");
    app->init();
    app->register_availability_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, on_availability);
    app->request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
    app->start();
}

I am using the provided CMakeLists.txt and .cpp file to run the vsomeip program in the WAMR runtime. During the compilation process, the files are being compiled successfully. However, I am encountering runtime errors. The specific errors are as follows: 

[14:30:49:118 - 7FA1BC4E5CC0]: warning: failed to link import function (env, _ZN10vsomeip_v37runtime3getEv)

TianlongLiang, 2023-07-04:
I suspect that this is because the name mangling of C++ is incompatible with WebAssembly. Could you dump the vsomeip library that you are linking against to look for the symbol name for vsomeip::runtime::get()? It should be something like `vsomeip_v3::runtime::get()`, if it does not exist, that may be the problem.
```shell
# to see which vsomeip shared library you are linking against
ldd main 
# in the build directory of vsomeip library, dump the symbol of that shared library
# it may not be exactly libvsomeip3.so, you can replace with the one you discovered before
nm libvsomeip3.so | grep "vsomeip_v3::runtime::get"
```

TianlongLiang, 2023-07-04:
The reason WAMR failed to link the `vsomeip::runtime::get()` is that your program is trying to directly use a dynamic library. You can either:

1. Create C wrapper APIs for the library native API, you can refer to this [sample](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/native-lib)
2. Alternatively, you can link the library statically. Assuming this library supports static linking. Assuming `vsomeip` supports static linking

I would recommend you method 2 if you just want to give it a try and this library would only be used in a small scope. If you want to use it in the production environment or it's for high scalable application, method 1 would be a better choice.

Anilkumar1818, 2023-08-28:
After creating the native library using the C wrapper function, we are encountering the same error: Failed to link import function.

TianlongLiang, 2023-08-28:
In the C wrapper, what function name are you setting? Is the mangling name like `_ZN10vsomeip_v37runtime3getEv`?

TianlongLiang, 2023-08-28:
Let's wind back a little, and make sure I understand your situation correctly:

At first, you have a CPP program that you posted initially, compiled it into wasm, run it with iwasm and the error is:
```
[14:30:49:118 - 7FA1BC4E5CC0]: warning: failed to link import function (env, _ZN10vsomeip_v37runtime3getEv)
```

Then, you write a C wrapper, similar to this:
```Cpp
#include <vsomeip/vsomeip.hpp>
#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678

std::shared_ptr< vsomeip::application > app;

void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available) {
std::cout << "Service ["
<< std::setw(4) << std::setfill('0') << std::hex << _service << "." << _instance
<< "] is " << (_is_available ? "available." : "NOT available.") << std::endl;
std::cout<<"hello";
}

extern "C"{
static int
vsomeip_main_wrapper(wasm_exec_env_t exec_env, ...)
{
    std::cout << "vsomeip appp \n";
    app = vsomeip::runtime::get()->create_application("Hello");
    app->init();
    app->register_availability_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, on_availability);
    app->request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
    app->start();
}
}

#define REG_NATIVE_FUNC(func_name, signature) \
    { #func_name, func_name##_wrapper, signature, NULL }

static NativeSymbol native_symbols[] = {
    REG_NATIVE_FUNC(vsomeip_main, "...")
};

uint32_t
get_native_lib(char **p_module_name, NativeSymbol **p_native_symbols)
{
    *p_module_name = "env";
    *p_native_symbols = native_symbols;
    return sizeof(native_symbols) / sizeof(NativeSymbol);
}
```
Now you have a C wrapper Cpp file.  
And you are still using a Cpp file for your main program to compile it to wasm, right?
```cpp

extern "C"{
int vsomeip_main(...);
}

int main(){
  vsomeip_main(...);
  ...;
}
```

The name mangling of Cpp is different from C, you may need to add the `extern "C"{}` block in the wrapper and in your main cpp program function definition. If you haven't tried that, that may be the problem.

Anilkumar1818, 2023-08-28:
Like Vsomeip in the same time we are trying the shared memory and message queue. F
or that also we are getting same errors Like Failed to link import function.

TianlongLiang, 2023-08-28:
It seems that your wrapper isn't used with the `extern "C"{}` block, you may want to try to use it to make your program use C's name mangling instead of C++ name mangling. And don't forget to use the block around your main program's wrapper definition

Anilkumar1818, 2023-08-28:
Thanks for the suggestions. Okay, let me give it a try.