wasm-micro-runtime/wasm-micro-runtime #2344: "WASM module instantiate failed: allocate memory failed" when using Alloc_With_Pool on x86-32

arvid-norlander, 2023-07-05:
Regardless of how large I set the heap size, when using x86-32, I get the error "WASM module instantiate failed: allocate memory failed". The exact same program (with the same build options otherwise) works on x86-64.

I'm using `wasm_runtime_full_init` as described in [the embedding documentation](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/embed_wamr.md). The issue arises when calling `wasm_runtime_instantiate`.

I don't yet have a minimal test case (that isn't tied into proprietary code) for this, but I can reproduce this with a very simple test WASM module (a single test function that takes and int32 and returns the same) compiled using WASI-SDK. If it would be useful, let me know and I can make a minimal C program that exhibits the issue.

I'm using WAMR-1.2.2 built as follows:
```cmake
set(WAMR_BUILD_PLATFORM "linux")

if(CMAKE_SYSTEM_PROCESSOR MATCHES "^(arm64|aarch64)")
    set(WAMR_BUILD_TARGET "AARCH64")
elseif(CMAKE_SYSTEM_PROCESSOR MATCHES "^(armv7a|cortex)")
    set(WAMR_BUILD_TARGET "ARMV7")
elseif(CMAKE_SYSTEM_PROCESSOR STREQUAL "x86_64")
    set(WAMR_BUILD_TARGET "X86_64")
    set(WAMR_BUILD_SIMD 1)
elseif(CMAKE_SYSTEM_PROCESSOR STREQUAL "x86")
    set(WAMR_BUILD_TARGET "X86_32")
else()
    message(SEND_ERROR "Unsupported build target platform ${CMAKE_SYSTEM_PROCESSOR}!")
endif()

set(WAMR_BUILD_INTERP 1)
set(WAMR_BUILD_FAST_INTERP 1)
set(WAMR_BUILD_AOT 1)
set(WAMR_BUILD_LIBC_BUILTIN 1)
set(WAMR_BUILD_LIBC_WASI 1)
set(WAMR_ROOT_DIR "${CMAKE_CURRENT_SOURCE_DIR}/wasm-micro-runtime")

include (${WAMR_ROOT_DIR}/build-scripts/runtime_lib.cmake)
add_library(vmlib ${WAMR_RUNTIME_LIB_SOURCE})
target_include_directories(vmlib PUBLIC ${WAMR_ROOT_DIR}/core/iwasm/include)
target_link_libraries (vmlib INTERFACE -lm -ldl -lpthread)
```

arvid-norlander, 2023-07-05:
It appears after some debugging that it tries to allocate 16 MB when I have set the pool to be 10 MB. Also for `wasm_runtime_instantiate` I'm passing a stack size of 2048 and same for the heap size. I don't see why it would try to allocate a whole 16 MB. This is more than I can afford on my 32-bit target.

arvid-norlander, 2023-07-05:
The issue goes away if I rebuild the test program with WASI-SDK instead of emscripten in standalone mode. Not sure what is going on, possibly an emscripten bug? Also why doesn't the issue happen in 64-bit builds of WAMR?

yamt, 2023-07-06:
can you share the wasm module which causes the problem?

arvid-norlander, 2023-07-06:
> can you share the wasm module which causes the problem?

Yes, this was a simple test module for a unit test of our C++ bindings for WAMR. In the attached archive is both the source code and the compiled module: [simple_wasm.tar.gz](https://github.com/bytecodealliance/wasm-micro-runtime/files/11965519/simple_wasm.tar.gz)

The build command was: `emcc -O3 simple_wasm.c -o simple_wasm_emcc.wasm -s STANDALONE_WASM`

If I instead build it with WASI-SDK, it works just fine: `/opt/wasi-sdk/bin/clang -O3 simple_wasm.c -o simple_wasm_wasisdk.wasm`

The strange thing is that it believes it need 16 MB of allocation, but only in the 32-bit build of WAMR. The 64-bit build of WAMR does not fail the memory allocation, even though the same memory pool with the same settings is used.

yamt, 2023-07-06:
thank you.

because the module has 16MB linear memory, 16MB allocation is normal.

depending on configurations, the linear memory is served by mmap. (ie. out of the configured memory pool.)
i guess it's what happens for your 64-bit case.

arvid-norlander, 2023-07-06:
Ah okay, thanks. I guess the issue is that emscripten has weird defaults then? I'm very new to wasm, so I hope you can forgive me for not understanding all the intricate details of it.

closed 2023-07-06 by arvid-norlander

yamt, 2023-07-06:
> Ah okay, thanks. I guess the issue is that emscripten has weird defaults then? I'm very new to wasm, so I hope you can forgive me for not understanding all the intricate details of it.

although i'm not familiar with emscripten, i guess it has some option to control the "initial-memory".