wasm-micro-runtime/wasm-micro-runtime #2345: Does not support C++exceptions, and does not support emcc exceptions? Are there any other solutions

w4454962, 2023-07-05:
There are abnormal requirements now, but none seem to have been implemented, even the "dev/exc_handling" branch has not fully seen the implementation of import functions when there are C++exceptions running.

woodsmc, 2023-07-06:
The dev/exc_handling branch is expected to have support for WASM exceptions, as per (this issue)[https://github.com/bytecodealliance/wasm-micro-runtime/issues/1884]. It's taken longer than excepted to get the implementation completed.

w4454962, 2023-07-08:
The temporary solution now is to encapsulate a security callback function, similar to "lua_pcall" and "lua_xpcall" in lua, to avoid code termination and exit.

woodsmc, 2023-07-13:
We hope to have an initial implementation of WASI exception handling coming to WAMR in the next few weeks.

There is still an open question about tool chain support. At the moment the Emscripten compiler tool chain supports WASI Exceptions for C++. The WASI SDK doesn't at the moment.