wasm-micro-runtime/wasm-micro-runtime #2351: Request for TensorFlow Lite Workload Update

JamesMenetrey, 2023-07-10:
Dear developers,

I've been working with the TensorFlow Lite workload found in your repository (https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/workload/tensorflow) which is currently using TensorFlow v2.4. Kudos to @wustwn for the original implementation.

Considering the benefits of newer versions of TensorFlow Lite, I wanted to inquire if there are any plans to update this workload to be compatible with a more recent version, either by the WAMR team or directly @wustwn?

Many thanks!

Cheers,
Jämes

lum1n0us, 2023-07-11:
thanks for asking. We'll take a look.

JamesMenetrey, 2023-07-28:
Hi lum1n0us,

Thanks for considering the TF-lite port upgrade. To provide more insights on why we would need this upgrade: we are evaluating multiple workloads for a scientific paper, with TF-lite being our top choice. However, the current version's missing features are a barrier to our benchmarking.

We truly appreciate your continued support and effort on this.

Cheers!

MMM-Jasmin, 2023-08-01:
Hi there!

I would also be really interested in an update to a newer Tensorflow version!
Some operations are missing and the version here is quite old now.

Any idea if it's feasible?

Kind regards,
Nils

lum1n0us, 2023-08-23:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/2369 is an ongoing PR. we've resolved all compilation problems. Now we are working on issues caused by "emscripten standalone supporting".

https://github.com/emscripten-core/emscripten/blob/main/system/lib/standalone/standalone.c#L139
and
https://github.com/emscripten-core/emscripten/blob/main/system/lib/standalone/standalone.c#L92.

We have a patch for `getentropy()` https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/workload/XNNPACK/xnnpack.patch#L64-L71

`openat` is ongoing. We plan to map it to WASI-IO system. It may take some time.

JamesMenetrey, 2023-08-26:
Hello @lum1n0us ,

This looks super great! Many thanks for your effort!
I'm looking forward to testing it when you think it's ready :)

Cheers

JamesMenetrey, 2023-10-06:
Hello @lum1n0us,

Were you able to resolve the challenges you encountered with Emscripten? I'm eager to try it out when I get the chance. :)

Thanks and Best regards!

lum1n0us, 2023-10-31:
@JamesMenetrey There is a workable [draft PR](https://github.com/bytecodealliance/wasm-micro-runtime/pull/2369). please take a look and let me know your opinion

g0djan, 2023-11-22:
@JamesMenetrey Hey could you take a look ^? It's just that maybe the fix can be merged and it's one less worry for WAMR-1.2.4 release

lum1n0us, 2023-12-14:
It seems `bazel build` didn't work as we expected.

```
WARNING: Download from https://storage.googleapis.com/mirror.tensorflow.org/github.com/llvm/llvm-project/archive/dc275fd03254d67d29cc70a5a0569acf24d2280d.tar.gz failed: class java.io.FileNotFoundException GET returned 404 Not Found
WARNING: Download from https://storage.googleapis.com/mirror.tensorflow.org/github.com/tensorflow/runtime/archive/7d879c8b161085a4374ea481b93a52adb19c0529.tar.gz failed: class java.io.FileNotFoundException GET returned 404 Not Found
ERROR: Skipping '//tensorflow/lite/tools/benchmark:benchmark_model-wasm': no such target '//tensorflow/lite/tools/benchmark:benchmark_model-wasm': target 'benchmark_model-wasm' not declared in package 'tensorflow/lite/tools/benchmark' (did you mean 'benchmark_model_main'?) defined by /home/ubuntu/dev/test_tflite_with_wamr/samples/workload/tensorflow/tensorflow/tensorflow/lite/tools/benchmark/BUILD
```

my guess is:
- if the patch was applied successfully? 
- if bazel downloaded all dependencies successfully?

in order to figure that out:
- goto tensorflow source directory. it should be */samples/workload/tensorflow/tensorflow*. use `git status` to check modification
- use `bazel query --output label_kind "//tensorflow/lite/..." | grep benchmark_model-wasm` to make sure bazel has all necessary