wasm-micro-runtime/wasm-micro-runtime #2356: Proposal for optimizing memory usage for read-only data

no1wudi, 2023-07-12:
# Motivation

* To reduce the memory usage for static typed languages (C/C++/Rust/Zig)

# Possible approaches

Currently, wasm doesn't have a concept of constant memory. That is,
all the memory is writable and will be loaded into the ram.

However, there are typical patterns where a module has a large amount
of read-only data. For example, a module might have a large array of
constant strings.

In such cases, it's possible to put the read-only data into a special
region of the linear memory by a linker script.

```
SECTIONS {
  . = 0x8;
  .rodata : {
    __rodata_start = .;
    *(.rodata)
    __rodata_end = .;
  }
  /* other sections */
  (...)
  __memory_end = .;
}

```
> Note: Don't start from 0x0 because 0x0 means NULL pointer in most
scenarios. Start from 0x8 for 8-byte alignment.

Then, for runtime, if __rodata_start and __rodata_end are found in the
module, the runtime should do the following (example in XIP mode):

* Allocate linear memory with the size of __memory_end - __rodata_end
* Don't need to copy the contents of __rodata_start and __rodata_end
* For memory accesses in the range of __rodata_start and __rodata_end,
  redirect them to the region of AOT file's memory section.
* For memory accesses out of the range of __rodata_start and __rodata_end,
  subtract __rodata_end from the address and then access the linear memory

If the module doesn't have __rodata_start and __rodata_end, the runtime treats
the module as a normal module.

If the module has __rodata_start and __rodata_end, but the runtime doesn't support
constant memory, the runtime treats the module as a normal module.

# Side effects

* Non-standard behavior for wasm
* Poor performance for memory accesses due to the extra address calculation

lum1n0us, 2023-07-18:
Just need to be sure:

- this is about adding a new section in the .aot file only. It won't affect .wasm files.
- the proposal wants to keep constant data in the file system only.
- there is a potential assumption that the file content won't be dropped during the .wasm running.

no1wudi, 2023-07-18:
> * this is about adding a new section in the .aot file only. It won't affect .wasm files.

It dont't introduce any new section to .aot/.wasm file, just add a `__heap_base` like symbol definition (__rodata_start  and __rodata_end ) by a special link script during wasm app compilation.

It's most useful for AOT XIP mode since the module is on ROM and can be accessed randomly. Thus we can save RAM for the constant memory. For interpreter mode, the .wasm file is already copied into RAM, so may not take benefit from it.

> * the proposal wants to keep constant data in the file system only.

Yes, keep them on read only storage.

> * there is a potential assumption that the file content won't be dropped during the .wasm running.

Yes, XIP file won't be dropped, if apply it to interpreter, memory section shouldn't be dropped at least.

lum1n0us, 2023-07-18:
Yes. No new section. Just new variables. 

Another question is, is there any difference between the read speed on both sides?

no1wudi, 2023-07-18:
> Another question is, is there any difference between the read speed on both sides?

Reading from ROM usually slower than RAM, but it's should be OK for most application that in native C application, they placed in ROM by default.

yangyalei, 2023-07-20:
> Another question is, is there any difference between the read speed on both sides?
I think I can give a test result after this done. after all, It make no difference if we don't add param to add this "__rodata_start"(by default?).

no1wudi, 2023-07-21:
Maybe we can achive this by custom section instead of change the runtime

yangyalei, 2023-07-25:
> Maybe we can achive this by custom section instead of change the runtime

What does it mean？ needen't add  **__rodata_start** and **__rodata_end**?

closed 2023-08-09 by no1wudi