wasm-micro-runtime/wasm-micro-runtime #2359: Set per-module constraints on instruction executions and memory usage

dzobbe, 2023-07-14:
Hello all, 

In a research project I am doing, I'd be interested in throttling the performance of wasm modules and add constraints in terms of memory usage. The latter could be of course achieved statically at build time by just giving X memory to the module. What about the first point? Would it be potentially doable to e.g. limit the number of wasm instructions that gets processed?

Thanks a lot in advance.

lum1n0us, 2023-07-17:
I am not sure if fully understand the task. Please correct me if I am in the wrong way. I guess it is about to limit every module instance resource(CPU and memory) usage and it doesn't include runtime consumption. 

In current phase, since a module instance can only access its linear memory, there is a very rough but very quick constraints by setting memory limitation in .wasm.

CPU usage will be another story. Maybe we can simplify the task by running a single module instance application and do some rough process constaints? By using start/end points, it is able to reduce interference of runtime.

dzobbe, 2023-07-18:
Yeah I don't care about the runtime but about the limitation on modules. And yes I got the memory one is a possibility. The CPU is a bit harder to understand maybe.

lum1n0us, 2023-07-18:
Can *tasket* or *docker* finish the job if only running one instance with one module?