wasm-micro-runtime/wasm-micro-runtime #2370: `wamrc --target=help/--cpu=help` etc doesn't work

arvid-norlander, 2023-07-19:
```console
$ wamrc --help
Usage: wamrc [options] -o output_file wasm_file
  --target=<arch-name>      Set the target arch, which has the general format: <arch><sub>
                            <arch> = x86_64, i386, aarch64, arm, thumb, xtensa, mips,
                                     riscv64, riscv32.
                              Default is host arch, e.g. x86_64
                            <sub> = for ex. on arm or thumb: v5, v6m, v7a, v7m, etc.
                            Use --target=help to list supported targets
  --target-abi=<abi>        Set the target ABI, e.g. gnu, eabi, gnueabihf, msvc, etc.
                              Default is gnu if target isn't riscv64 or riscv32
                              For target riscv64 and riscv32, default is lp64d and ilp32d
                            Use --target-abi=help to list all the ABI supported
  --cpu=<cpu>               Set the target CPU (default: host CPU, e.g. skylake)
                            Use --cpu=help to list all the CPU supported
  --cpu-features=<features> Enable or disable the CPU features
                            Use +feature to enable a feature, or -feature to disable it
                            For example, --cpu-features=+feature1,-feature2
                            Use --cpu-features=+help to list all the features supported
[...]
```

This indicates it should be possible to get lists of supported targets etc. However this doesn't work, for example:

```console
$ wamrc --target=help
Usage: wamrc [options] -o output_file wasm_file
[... same help as before ...]
```

```console
$ wamrc --version
wamrc 1.2.2
```

xujuntwt95329, 2023-07-19:
Hi @arvid-norlander, thanks for reporting this issue 

Seems the current logic for processing `--target=help` option is not friendly, you need to use `wamrc --target=help -o <aot_file> <wasm_file>`, the `-o <aot_file> <wasm_file>` can't be omitted even if they are not used.

And for --cpu=help:
`wamrc --target=xxx --cpu=help -o <aot_file> <wasm_file>`

We will enhance this processing logic to make it more friendly.

arvid-norlander, 2023-07-19:
> Hi @arvid-norlander, thanks for reporting this issue
> 
> Seems the current logic for processing `--target=help` option is not friendly, you need to use `wamrc --target=help -o <aot_file> <wasm_file>`, the `-o <aot_file> <wasm_file>` can't be omitted even if they are not used.

Even further, the wasm-file has to actually exist as well!

> 
> And for --cpu=help: `wamrc --target=xxx --cpu=help -o <aot_file> <wasm_file>`
> 
> We will enhance this processing logic to make it more friendly.

xujuntwt95329, 2023-07-19:
Yes, currently it is really not friendly to users, we have planned to fix this.