wasm-micro-runtime/wasm-micro-runtime #2391: Cross-compile iwasm for AARCH64 on x86: fatal error: arm_neon.h: No such file or directory

YogzZ, 2023-07-26:
I want to cross-compile the iwasm for aarch64 on x86, but got follow error:
```shell
wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:4216:10: fatal error: arm_neon.h: No such file or directory
 #include <arm_neon.h>
          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/iwasm_shared.dir/build.make:468: CMakeFiles/iwasm_shared.dir/core/iwasm/common/wasm_runtime_common.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:85: CMakeFiles/iwasm_shared.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

I use the following instruction:
```shell
mkdir wasm-micro-runtime/build
cd wasm-micro-runtime/build (main)
cmake .. -DWAMR_BUILD_PLATFORM=linux -DWAMR_BUILD_TARGET=AARCH64
make clean
make
```

What's the problem with my operation?

TianlongLiang, 2023-08-03:
Hi, are you using regular gcc/clang instead of those support cross-compile? If so, you can try to install and use `gcc-arm-linux-gnueabihf`:
```shell
# to install
sudo apt-get install gcc-arm-linux-gnueabihf
# to use with cmake
CC=arm-linux-gnueabihf-gc cmake .. -DWAMR_BUILD_PLATFORM=linux -DWAMR_BUILD_TARGET=AARCH64
```

TianlongLiang, 2023-08-03:
Hi, are you using regular gcc/clang instead of those support cross-compile? If so, you can try to install and use `gcc-aarch64-linux-gnu`:
```shell
# to install
sudo apt-get update && sudo apt-get install gcc-aarch64-linux-gnu
# to use with cmake
CC=aarch64-linux-gnu-gcc cmake -DWAMR_BUILD_PLATFORM=linux -DWAMR_BUILD_TARGET=AARCH64 -DWAMR_BUILD_SIMD=0 -B build
```