wasm-micro-runtime/wasm-micro-runtime #2396: Violate witx spec while executing the given wasm file

luxinyi0105, 2023-07-27:
## Description

wasm-micro-runtime may not be compliant with the witx spec while executing the given test case.

Actually, the given test case was simply mutated a wasm file, which was obtained by compiling the C program generated with Csmith using the Emscripten compiler (Emcc). You can convert wasm file to wat format with wabt and find the mutation operation in line 9974(replace `i32.sub` with `i32.div_s`). Releated files is [here](https://github.com/luxinyi0105/wasm_runtime_test/raw/main/07-26.zip).

## Current State

```sh
$ iwasm mutated.wasm 
checksum = DAFC642C
```

## Expected

As mentioned above, the mutation there is changing a subtraction of the stack pointer into a division of the stack pointer which means that the wasm module's in-memory shadow-stack pointer is getting corrupted. Notably what this causes is the stack to get misaligned.

[witx spec](https://github.com/WebAssembly/wasi/blob/main/legacy/tools/witx-docs.md#pointers) says `misaligned pointers ... shall trap.` It seems that WAMR needs to return a trap while meeting this situatuion, but actually it correctly output the checksum result and did not trigger any abnormal behavior.

## Environment

- Hardware Architecture: x86_64
- Operating system: Ubuntu 22.04
- Tools Version: iwasm 1.2.2

wenyongh, 2023-07-31:
@luxinyi0105 Not sure what the `misaligned pointers` means? The `witx spec` that you mentioned seems to use `pointer` keyword to describe the function arguments:
```wast
   (param $in (@witx const_pointer $subscription))
   (param $out (@witx pointer $event))
```
But the wasm/wast file you uploaded doesn't use `pointer` keyword in the function declaration:
```wast
  (func (;23;) (type 1) (param i32 i32 i32) (result i32)
    (local i32 i32 i32 i32 i32 i32 i32)
    global.get 0
    i32.const 32
    i32.div_s
```
In fact, it isn't supported by WAMR even it is added. The running result is same with several other runtimes, e.g. wasmer and wavm.

luxinyi0105, 2023-08-07:
> In fact, it isn't supported by WAMR even it is added. The running result is same with several other runtimes, e.g. wasmer and wavm.

Thanks for your reply. But while execute the given testcase with wasmer, you will find `"Aborted (core dumped)"`. In this sense, the running result is not the same.

```sh
$ wasmer mutated.wasm
checksum = DAFC642C
thread '<unnamed>' panicked at 'misaligned pointer dereference: address must be a multiple of 0x4 but is 0x7f36643857f9', /home/oscar/Software/wasmer/lib/api/src/sys/mem_access.rs:63:13
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread caused non-unwinding panic. aborting.
Aborted (core dumped)
```
We believe that wasmer's output is not entirely correct, it would be better to trigger wasmer's internal exception handling structures and output `"Runtime Error"` instead of the current `"Aborted"`. But it at least illustrates the problem of `misaligned pointer` in the given testcase, while WAMR does not.

Due to a lack of professional understanding, I am not able to answer your question about `misaligned pointers`, perhaps wasmer's error message can inspire you.