wasm-micro-runtime/wasm-micro-runtime #2401: An interesting crash report

erxiaozhou, 2023-07-28:
## Test case
[select_ast_4871_26_16832064820804524.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12196853/select_ast_4871_26_16832064820804524.zip)

## Build commands
I compile the code with commit id 7db4815e837bc34942e95aba292ca0eca318cc54.
Platform: Ubuntu 20.04
CPU: amd64 

## compile:
```shell
export CC=/usr/bin/clang-14
export CXX=/usr/bin/clang++-14
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWASM_ENABLE_SIMD=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Release -DWAMR_BUILD_FAST_INTERP=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -Bbuild
cd build
make -j40
```
## execute:
```shell
iwasm --heap-size=0 -f to_test <tc_name>
```

## Behavior:
I used the fast interpreter to execute the test case repeatedly 11 times, two of which triggered a crash. It is worth mentioning that the classic interpreter will not crash when executing this test case.

Crash 
```
Segmentation fault (core dumped)
```
**The behavior of the fast interpreter**
[image]

**The behavior of the classic interpreter**
[image]

xujuntwt95329, 2023-08-17:
@erxiaozhou   Thanks for reporting the interesting issue, this has been fixed in https://github.com/bytecodealliance/wasm-micro-runtime/pull/2476

erxiaozhou, 2023-08-17:
Thank you for your reply!