wasm-micro-runtime/wasm-micro-runtime #2412: LLVM ERROR: failed to perform tail call elimination with MIPS AOT

eloparco, 2023-07-31:
Hello, I'm getting this error when trying to compile for MIPS:

```bash
$ wasm-micro-runtime/wamr-compiler/build/wamrc --enable-multi-thread --disable-simd --target=mips --size-level=1 -o pigz.aot pigz.wasm
Create AoT compiler with:
  target:        mips
  target cpu:    
  target triple: mips-pc-linux-gnu
  cpu features:  
  opt level:     3
  size level:    1
  output format: AoT file
LLVM ERROR: failed to perform tail call elimination on a call site marked musttail
78805 Abort trap: 6           wasm-micro-runtime/wamr-compiler/build/wamrc --enable-multi-thread --disable-simd --target=mips --size-level=1 -o "pigz.aot" "pigz.wasm"
```

The wasm file is [pigz.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12216660/pigz.wasm.zip) and it's coming from https://github.com/bytecodealliance/wasm-parallel-gzip.
It works fine with other targets (I tried aarch64, armv7, x86_64).

eloparco, 2023-08-01:
I just checked out version `1.2.2` and there the compilation is fine, so the issue was introduced after that.

loganek, 2023-08-01:
I just did a quick git bisect check; looks like the PR #2244 caused the regression.

yamt, 2023-08-14:
i tested https://github.com/bytecodealliance/wasm-micro-runtime/pull/2457 with the given wasm file lightly.
(lightly == just aot compile. i have no mips env.)

closed 2023-08-15 by wenyongh