wasm-micro-runtime/wasm-micro-runtime #2425: failed to call unlinked import function (env, console.log)

lightbluewind, 2023-08-05:
Hi Master,

I've built the WAMR in Android, I can run wasm application compiled from C code.

I wrote a simple test code in Assemblyscript:

```
import "console";

console.log("Webassembly initialized");
```

I run the compiled test.wasm in Android environment, got an error:
"failed to call unlinked import function (env, console.log)"

Does this mean WAMR doesn't fully support Assemblyscript standard object "console"?

tonibofarull, 2023-08-05:
You have to use the wasi layer for AS: https://github.com/jedisct1/as-wasi
The symbol `console.log` will be translated to the symbols of wasi-snapshot-preview-1.

lightbluewind, 2023-08-06:
Thanks @tonibofarull 

I used "as-wasi", Assemblyscript code is like:

```
// Import from the installed as-wasi package
import { Console as console1, Environ } from "as-wasi/assembly";

// Create an environ instance
let env = new Environ();

// Get the HOME Environment variable
let home = env.get("HOME")!;

// Log the HOME string to stdout
console1.log(home);
console1.log("Hi, this is an Assemblyscript demo");
```

> However, I got error:
> 
> wasm_runtime_instantiate
> 
>  warning: failed to link import function (env, abort)
> 
>  warning: failed to link import function (env, seed)
> 
> WASM module instantiate failed: Exception: failed to call unlinked import function (env, abort)

Then I tried to follow this #510 , I failed to apply the patch, seems the patch was based on old commits.

How could I fix this issue?

tonibofarull, 2023-08-10:
I haven't used Environ before, can you test compiling a simple script with only the following?
```as
import {Console} from "as-wasi/assembly";
Console.log("hello world!");
```

Can you share the full code + configuration, please?