wasm-micro-runtime/wasm-micro-runtime #2430: wasi-nn + threads

yamt, 2023-08-07:
wasm_deinstantiate and aot_deinstantiate call wasi_nn_destroy only for !is_sub_inst.

on the other hand, wasm_runtime_get_wasi_nn_ctx seems to allocate the context for each instances.

probaby it just means that (this implementation of?) wasi-nn is not compatible with threads.

tonibofarull, 2023-08-07:
For the wasi-nn we have a new (not yet PR'd) implementation where the context is managed within the library.
The motivations are the following:
1. Allowing the user to build is as a native library.
2. Avoid coupling with WAMR as much as possible.

yamt, 2023-08-08:
> What would be required to support it with threads?

i don't know.
probably
* share the context among instances. (wasi-threads and lib-pthread are using instance per thread model)
* add some locks