wasm-micro-runtime/wasm-micro-runtime #2432: Run wasm file fail

yangyalei, 2023-08-07:
Env:  Ubuntu-20.04  x86_64

Here is my stage:
1.sudo apt install build-essential cmake g++-multilib libgcc-9-dev lib32gcc-9-dev ccache
2.cd product-mini/platforms/linux/
mkdir build && cd build
cmake ..
make
3. add test.c 
```
#include <stdio.h>
#include <stdlib.h>

int main(int argc, char **argv)
{
    char *buf;

    printf("Hello world!\n");

    buf = malloc(1024);
    if (!buf) {
        printf("malloc buf failed\n");
        return -1;
    }

    printf("buf ptr: %p\n", buf);

    sprintf(buf, "%s", "1234\n");
    printf("buf: %s", buf);

    free(buf);
    return 0;
}
```
4./opt/wasi-sdk/bin/clang -O3 -o test.wasm test.c  ///opt/wasi-sdk/bin/clang download from https://github.com/WebAssembly/wasi-sdk/releases/download/wasi-sdk-20/wasi-sdk-20.0-linux.tar.gz
5. ubuntu:./iwasm  test.wasm
**非法指令 (核心已转储)**

Any way to figure it out ?

yangyalei, 2023-08-07:
严格按照官方文档来操作的，谁帮忙确认是哪里的问题？

wenyongh, 2023-08-08:
Hi, could you try building iwasm by adding `-DWAMR_DISABLE_WRITE_GS_BASE=1` to cmake?

yangyalei, 2023-08-08:
> Hi, could you try building iwasm by adding `-DWAMR_DISABLE_WRITE_GS_BASE=1` to cmake?

Happy to cry. It works.
Can We update this to documents, It waste me a lot of time.

wenyongh, 2023-08-08:
I submit a PR to auto-check the feature in cmake script, could you check whether it works for you? Or, rebuild iwasm again with it?

yangyalei, 2023-08-08:
> I submit PR #2437 to auto-check the feature in cmake script, could you check whether it works for you? Or, rebuild iwasm again with it?

Test ok.

closed 2023-08-08 by yangyalei