wasm-micro-runtime/wasm-micro-runtime #2440: Run nostdlib wasm app fail

yangyalei, 2023-08-08:
Env: Ubuntu 20.04 X86_64 

Build:
```
wasm-micro-runtime/product-mini/app-samples/hello-world$ cat build.sh 
# Copyright (C) 2019 Intel Corporation.  All rights reserved.
# SPDX-License-Identifier: Apache-2.0 WITH LLVM-exception

WAMR_DIR=${PWD}/../../..

echo "Build wasm app .."
/opt/wasi-sdk/bin/clang -O3 \
        -z stack-size=4096 -Wl,--initial-memory=65536 \
        -o test.wasm main.c \
        -Wl,--export=main -Wl,--export=__main_argc_argv \
        -Wl,--export=__data_end -Wl,--export=__heap_base \
        -Wl,--strip-all,--no-entry \
        -Wl,--allow-undefined \
        -nostdlib \

echo "Build binarydump tool .."
rm -fr build && mkdir build && cd build
cmake ../../../../test-tools/binarydump-tool
make
cd ..

echo "Generate test_wasm.h .."
./build/binarydump -o test_wasm.h -n wasm_test_file test.wasm

echo "Done"
$
$ ./build.sh 
Build wasm app ..
Build binarydump tool ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mi/code/wasm-micro-runtime/product-mini/app-samples/hello-world/build
Scanning dependencies of target binarydump
[ 50%] Building C object CMakeFiles/binarydump.dir/binarydump.c.o
[100%] Linking C executable binarydump
[100%] Built target binarydump
Generate test_wasm.h ..
Done

```
Run:
 ```
$ ./iwasm -v=5 test.wasm 
[12:06:57:879 - 7FBC819B7CC0]: Load type section success.

[12:06:57:893 - 7FBC819B7CC0]: Load import section success.

[12:06:57:895 - 7FBC819B7CC0]: Load function section success.

[12:06:57:897 - 7FBC819B7CC0]: Load memory section success.

[12:06:57:899 - 7FBC819B7CC0]: Load global section success.

[12:06:57:901 - 7FBC819B7CC0]: Load export section success.

[12:06:57:902 - 7FBC819B7CC0]: Load code segment section success.

[12:06:57:904 - 7FBC819B7CC0]: Load data segment section success.

[12:06:57:905 - 7FBC819B7CC0]: Found aux __data_end global, value: 1082
[12:06:57:906 - 7FBC819B7CC0]: Found aux __heap_base global, value: 5184
[12:06:57:907 - 7FBC819B7CC0]: Found aux stack top global, value: 5184, global index: 0, stack size: 4096
[12:06:57:917 - 7FBC819B7CC0]: Shrink memory size to 5184
[12:06:57:919 - 7FBC819B7CC0]: Load module success.

[12:06:57:922 - 7FBC819B7CC0]: Memory instantiate:
[12:06:57:924 - 7FBC819B7CC0]:   page bytes: 5184, init pages: 1, max pages: 1
[12:06:57:925 - 7FBC819B7CC0]:   heap offset: 5184, heap size: 0

[12:06:57:932 - 7FBC819B7CC0]: Memory instantiate success.
[12:06:57:965 - 7FBC819B7CC0]: warning: allocate 16 bytes memory failed
Exception: allocate memory failed
```

no1wudi, 2023-08-09:
@yangyalei Please notice that your `heap size` is 0, did your enabled libc-wasi for your `iwasm` build ? if that, try to launch this module with explict heap setting like `./iwasm --heap-size=16384 -v=5 test.wasm`

yangyalei, 2023-08-09:
> @yangyalei Please notice that your `heap size` is 0, did you enabled libc-wasi for your `iwasm` build ? if that, try to launch this module with explict heap setting like `./iwasm --heap-size=16384 -v=5 test.wasm`

Is the --heap-size param must be add when run wasm? I have enabled libc-wasi and libc-builtin.

By the way, if build wasm app with -nostdlib param, Is it just relay on **libc-builtin** in `iwasm` build? The difference between  **libc-builtin** and **libc-wasi** really confused。

no1wudi, 2023-08-09:
Only needed if libc-wasi enabled and you want to run application with libc-builitn, please see: https://github.com/bytecodealliance/wasm-micro-runtime/pull/2346

`libc-builtin` and libc-wasi is totally different, libc-builtin provide the libc API directly that implemented by WAMR itself, `libc-wasi` provide the wasi API to support wasi-libc and other wasi based ecosystems (Rust/Zig etc), it provide the API like:
```c
    REG_NATIVE_FUNC(args_get, "(**)i"),
    REG_NATIVE_FUNC(args_sizes_get, "(**)i"),
    REG_NATIVE_FUNC(clock_res_get, "(i*)i"),
    REG_NATIVE_FUNC(clock_time_get, "(iI*)i"),
    REG_NATIVE_FUNC(environ_get, "(**)i"),
    REG_NATIVE_FUNC(environ_sizes_get, "(**)i"),
    REG_NATIVE_FUNC(fd_prestat_get, "(i*)i"),
    REG_NATIVE_FUNC(fd_prestat_dir_name, "(i*~)i"),
    REG_NATIVE_FUNC(fd_close, "(i)i"),
    REG_NATIVE_FUNC(fd_datasync, "(i)i"),
    REG_NATIVE_FUNC(fd_pread, "(i*iI*)i"),
    REG_NATIVE_FUNC(fd_pwrite, "(i*iI*)i"),
    REG_NATIVE_FUNC(fd_read, "(i*i*)i"),
    REG_NATIVE_FUNC(fd_renumber, "(ii)i"),
    REG_NATIVE_FUNC(fd_seek, "(iIi*)i"),
    REG_NATIVE_FUNC(fd_tell, "(i*)i"),
    REG_NATIVE_FUNC(fd_fdstat_get, "(i*)i"),
    REG_NATIVE_FUNC(fd_fdstat_set_flags, "(ii)i"),
    REG_NATIVE_FUNC(fd_fdstat_set_rights, "(iII)i"),
    REG_NATIVE_FUNC(fd_sync, "(i)i"),
    REG_NATIVE_FUNC(fd_write, "(i*i*)i"),
    REG_NATIVE_FUNC(fd_advise, "(iIIi)i"),
    REG_NATIVE_FUNC(fd_allocate, "(iII)i"),
    REG_NATIVE_FUNC(path_create_directory, "(i*~)i"),
    REG_NATIVE_FUNC(path_link, "(ii*~i*~)i"),
    REG_NATIVE_FUNC(path_open, "(ii*~iIIi*)i"),
    REG_NATIVE_FUNC(fd_readdir, "(i*~I*)i"),
    REG_NATIVE_FUNC(path_readlink, "(i*~*~*)i"),
    REG_NATIVE_FUNC(path_rename, "(i*~i*~)i"),
    REG_NATIVE_FUNC(fd_filestat_get, "(i*)i"),
    REG_NATIVE_FUNC(fd_filestat_set_times, "(iIIi)i"),
    REG_NATIVE_FUNC(fd_filestat_set_size, "(iI)i"),
    REG_NATIVE_FUNC(path_filestat_get, "(ii*~*)i"),
    REG_NATIVE_FUNC(path_filestat_set_times, "(ii*~IIi)i"),
    REG_NATIVE_FUNC(path_symlink, "(*~i*~)i"),
    REG_NATIVE_FUNC(path_unlink_file, "(i*~)i"),
    REG_NATIVE_FUNC(path_remove_directory, "(i*~)i"),
    REG_NATIVE_FUNC(poll_oneoff, "(**i*)i"),
    REG_NATIVE_FUNC(proc_exit, "(i)"),
    REG_NATIVE_FUNC(proc_raise, "(i)i"),
    REG_NATIVE_FUNC(random_get, "(*~)i"),
    REG_NATIVE_FUNC(sock_accept, "(ii*)i"),
    REG_NATIVE_FUNC(sock_addr_local, "(i*)i"),
    REG_NATIVE_FUNC(sock_addr_remote, "(i*)i"),
    REG_NATIVE_FUNC(sock_addr_resolve, "($$**i*)i"),
    REG_NATIVE_FUNC(sock_bind, "(i*)i"),
    REG_NATIVE_FUNC(sock_close, "(i)i"),
    REG_NATIVE_FUNC(sock_connect, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_broadcast, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_keep_alive, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_linger, "(i**)i"),
    REG_NATIVE_FUNC(sock_get_recv_buf_size, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_recv_timeout, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_reuse_addr, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_reuse_port, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_send_buf_size, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_send_timeout, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_tcp_fastopen_connect, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_tcp_keep_idle, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_tcp_keep_intvl, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_tcp_no_delay, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_tcp_quick_ack, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_ip_multicast_loop, "(ii*)i"),
    REG_NATIVE_FUNC(sock_get_ip_multicast_ttl, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_ip_ttl, "(i*)i"),
    REG_NATIVE_FUNC(sock_get_ipv6_only, "(i*)i"),
    REG_NATIVE_FUNC(sock_listen, "(ii)i"),
    REG_NATIVE_FUNC(sock_open, "(iii*)i"),
    REG_NATIVE_FUNC(sock_recv, "(i*ii**)i"),
    REG_NATIVE_FUNC(sock_recv_from, "(i*ii**)i"),
    REG_NATIVE_FUNC(sock_send, "(i*ii*)i"),
    REG_NATIVE_FUNC(sock_send_to, "(i*ii**)i"),
    REG_NATIVE_FUNC(sock_set_broadcast, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_keep_alive, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_linger, "(iii)i"),
    REG_NATIVE_FUNC(sock_set_recv_buf_size, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_recv_timeout, "(iI)i"),
    REG_NATIVE_FUNC(sock_set_reuse_addr, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_reuse_port, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_send_buf_size, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_send_timeout, "(iI)i"),
    REG_NATIVE_FUNC(sock_set_tcp_fastopen_connect, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_tcp_keep_idle, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_tcp_keep_intvl, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_tcp_no_delay, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_tcp_quick_ack, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_ip_multicast_loop, "(iii)i"),
    REG_NATIVE_FUNC(sock_set_ip_multicast_ttl, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_ip_add_membership, "(i*i)i"),
    REG_NATIVE_FUNC(sock_set_ip_drop_membership, "(i*i)i"),
    REG_NATIVE_FUNC(sock_set_ip_ttl, "(ii)i"),
    REG_NATIVE_FUNC(sock_set_ipv6_only, "(ii)i"),
    REG_NATIVE_FUNC(sock_shutdown, "(ii)i"),
    REG_NATIVE_FUNC(sched_yield, "()i"),
```

yangyalei, 2023-08-09:
> Only needed if libc-wasi enabled and you want to run application with libc-builitn, please see: #2346
>
> `libc-builtin` and libc-wasi is totally different, libc-builtin provide the libc API directly [...]

Thanks，I got it.

closed 2023-08-09 by yangyalei

littetimo567, 2023-08-18:
Why is it different from the link description of https://wamr.gitbook.io/document/basics/getting-started/on_host?