wasm-micro-runtime/wasm-micro-runtime #2444: Zephyr output

andy9a9, 2023-08-10:
Dear community!

I would like to start using the Wasm project on Zephyr. So I followed the guides to compile and use the `simple-product`. Not sure if this is expected output or not. MCU: `stm32f756zg`

So what I did...
```sh
cd /home/vboxuser/zephyrproject/wasm/product-mini/platforms/zephyr/simple
cp boards/nucleo767zi.conf boards/nucleo765zg.conf
ln -s boards/nucleo765zg.conf prj.conf

west build . -d build -DWAMR_BUILD_TARGET=THUMBV7
```
Compilation seems to be ok
```
[0/1] Re-running CMake...
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/vboxuser/zephyrproject/wasm/product-mini/platforms/zephyr/simple
-- CMake version: 3.27.1
-- Cache files will be written to: /home/vboxuser/.cache/zephyr
-- Zephyr version: 3.4.0-rc1 (/home/vboxuser/zephyrproject/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.14.0")
-- Board: nucleo_f756zg
-- Found host-tools: zephyr 0.16.1 (/opt/zephyr-sdk-0.16.1)
-- Found toolchain: zephyr 0.16.1 (/opt/zephyr-sdk-0.16.1)
-- Found BOARD.dts: /home/vboxuser/zephyrproject/zephyr/boards/arm/nucleo_f756zg/nucleo_f756zg.dts
-- Generated zephyr.dts: /home/vboxuser/zephyrproject/wasm/product-mini/platforms/zephyr/simple/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/vboxuser/zephyrproject/wasm/product-mini/platforms/zephyr/simple/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/vboxuser/zephyrproject/wasm/product-mini/platforms/zephyr/simple/build/zephyr/dts.cmake
Parsing /home/vboxuser/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/vboxuser/zephyrproject/wasm/product-mini/platforms/zephyr/simple/build/zephyr/.config'
No change to configuration in '/home/vboxuser/zephyrproject/wasm/product-mini/platforms/zephyr/simple/build/zephyr/.config'
No change to Kconfig header in '/home/vboxuser/zephyrproject/wasm/product-mini/platforms/zephyr/simple/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Build Configurations:
     Build as target THUMBV7
     CMAKE_BUILD_TYPE 
     WAMR Interpreter enabled
     WAMR AOT enabled
     WAMR Fast JIT disabled
     WAMR LLVM ORC JIT disabled
     Libc builtin enabled
     Libc WASI disabled
     Fast interpreter disabled
     Multiple modules disabled
     Bulk memory feature enabled
     Reference types disabled
     Global heap pool enabled
     Custom global heap size: 131072
-- Configuring done (5.4s)
-- Generating done (0.1s)
-- Build files have been written to: /home/vboxuser/zephyrproject/wasm/product-mini/platforms/zephyr/simple/build
[6/14] Linking C executable zephyr/zephyr_pre0.elf

[9/14] Linking C executable zephyr/zephyr_pre1.elf

[14/14] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      116224 B         1 MB     11.08%
             RAM:      144832 B       256 KB     55.25%
            QSPI:          0 GB       256 MB      0.00%
            DTCM:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
```
Then I flashed the binary
```
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/vboxuser/zephyrproject/wasm/product-mini/platforms/zephyr/simple/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.11.0+dev-00726-ge2b6f5655 (2023-05-11-16:58)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
Info : clock speed 2000 kHz
Info : STLINK V2J40M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.236149
Info : [stm32f7x.cpu] Cortex-M7 r0p1 processor detected
Warn : [stm32f7x.cpu] Silicon bug: single stepping may enter pending exception handler!
Info : [stm32f7x.cpu] target has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f7x.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f7x.cpu       hla_target little stm32f7x.cpu       running

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08011b04 msp: 0x20033580
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : device id = 0x10016449
Info : flash size = 1024 kbytes
auto erase enabled
wrote 131072 bytes from file /home/vboxuser/zephyrproject/wasm/product-mini/platforms/zephyr/simple/build/zephyr/zephyr.hex in 4.250750s (30.112 KiB/s)

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
shutdown command invoked
```
Serial output from eval board looks like this
```
tr--- 19 messages dropped ---
: 0x1458
buf: 1234
elpase: 
```
If I compiled the `hello-world` on the host, it looks like this:
```
Hello world!
buf ptr: 0x1460
buf: 1234
```
I changed some bytes in `test_wasm.h`, but the output is still the same. So the question is if this is an expected behavior or did I something wrong?

Thanks
Andy

TianlongLiang, 2023-08-10:
Hi, I am getting this behavior(jitted output) too on my host Ubuntu machine, however, it's gone when I use Zephyr inside the docker container(run on qemu x86):

```
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.3.0  ***
Hello world!
buf ptr: 0x1458
buf: 1234
elpase: 0
QEMU: Terminated
root@bef1703f7456:~/wasm-micro-runtime/product-mini/platforms/zephyr/simple# 
```

Can you try to use docker to see whether this problem persists? 

I am using the `Dockerfile.old`, you can refer to [this document](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/product-mini/platforms/zephyr/simple#readme) for How to build Zephyr docker for WAMR

andy9a9, 2023-08-14:
Not sure, if this will help me, while there are some `esp` changes which are not related to my board.

On the other hand, how can I try an another "example" to see if there is some output change?

closed 2023-08-15 by andy9a9