wasm-micro-runtime/wasm-micro-runtime #2446: Do not terminate while executing the given testcase

luxinyi0105, 2023-08-10:
## Description

While executing [the given testcase](https://github.com/luxinyi0105/wasm_runtime_test/raw/main/mutated_file_28840.wasm), WAMR hangs out and does not end.

### Current State

```sh
$ iwasm mutated_file_28840.wasm 
```
It will hang out and does not end.

### Expected

I tried to use some other wasm runtimes, such as wasmer, wasmtime, and wasmi, to execute the same testcase. 

Wasmer will terminate abnormally due to "misaligned pointer", while wasmtime and wasmi are running normally but will not output anything.

Due to a lack of understanding of WebAssembly, I am unable to determine what the expected result is, maybe one of the two above.
 
## Environment
- Hardware Architecture: x86_64
- Operating system: Ubuntu 22.04
- Tools Version: iwasm 1.2.2

## Extra info

By the time I submitted this issue, it had been running for about 3 hours but still did not terminate.

wenyongh, 2023-08-15:
Hi, I tested with wasmer latest version, it also did not terminate. Not sure which version of wasmer did you use?

luxinyi0105, 2023-08-15:
> Hi, I tested with wasmer latest version, it also did not terminate. Not sure which version of wasmer did you use?

[image]

luxinyi0105, 2023-08-18:
> Hi, the wasmer version I installed is 4.1.1.

[image]

wenyongh, 2023-08-23:
It is a little strange that wasmer 4.1.1 hangs in my machine:
[image]

luxinyi0105, 2023-08-31:
> Wasmer will terminate abnormally due to "misaligned pointer", while wasmtime and wasmi are running normally but will not output anything.

Setting aside the different results obtained by using wasmer to execute the given test case, results of wasmtime and wasmi to execute the same test case are as follows:
[image]
Wasmi terminates normally without outputing anything. Wasmtime used to produce the same results as Wasmi, but the latest version of the tool was modified to produce the same results as Wasmer: terminate abnormally due to "misaligned pointer". In either case, it seems that the result of WAMR should not be "hang out".