wasm-micro-runtime/wasm-micro-runtime #2453: WAMRC is working bad for non x86-64 platforms

Zzzabiyaka, 2023-08-11:
I'm compiling the test I'm creating in #2411 to AOT and it fails with 

```
AOT module load failed: resolve symbol .aot_stack_sizes failed
 wasm_module_new failed
Error creating WAMR module: 
Module initialisation failed
1) Running spawn_stress_test_main_wamrc.aot
```

for wamrc built from 1.2.2 release it works ok. 

cc @wenyongh  for visibility.

yamt, 2023-08-14:
> I'm compiling the test I'm creating in #2411 to AOT and it fails with
>
> ```
> AOT module load failed: resolve symbol .aot_stack_sizes failed
>  wasm_module_new failed
> Error creating WAMR module:
> Module [...]

can you provide a simpler instruction to reproduce?
eg. a wasm module + wamr cli options

yamt, 2023-08-14:
which exact versions of iwasm and wamrc are you using?

do they both have https://github.com/bytecodealliance/wasm-micro-runtime/pull/2317 ? (i remember i tested it with qemu armhf. which "ARM" variant is yours?)

Zzzabiyaka, 2023-08-14:
Hi 

That's how we use wamrc

```
echo "Generating ARMv7 eabi AOT binary"
$WAMRC \
    --disable-simd \
    --target=armv7 \
    --target-abi=eabi \
    --cpu-features=-neon \
    --cpu=generic \
    --size-level=1 \
    -o "$SCRIPT_DIR/filename.aot" \
    "$SCRIPT_DIR/filename.wasm"
```

We use armv7 + aarch64. Both for eabi

Zzzabiyaka, 2023-08-14:
>which exact versions of iwasm and wamrc are you using?

latest main 

1.2.2 release shouldn't have #2317

Zzzabiyaka, 2023-08-14:
here is C file from which I generate wasm 

https://gist.github.com/Zzzabiyaka/e3879ee0d746c128aee94c757a80540c

Wasm is generated using 

```
$CC \
        -target wasm32-wasi-threads \
        -pthread -ftls-model=local-exec \
        -z stack-size=32768 \
        -Wl,--export=__heap_base \
        -Wl,--export=__data_end \
        -Wl,--shared-memory,--max-memory=1966080 \
        -Wl,--export=wasi_thread_start \
        -Wl,--export=malloc \
        -Wl,--export=free \
        -I $WAMR_DIR/samples/wasi-threads/wasm-apps \
        $test_c -o $test_wasm
 ```
 
 (https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/libraries/lib-wasi-threads/test/build.sh)
 
 aot is generated from wasm using the command above

yamt, 2023-08-14:
@Zzzabiyaka 

thank you for providing info.
i tried to reproduce the problem. but i failed.

my first attempt ended up with a seemingly unrelated error:
```
AOT module load failed: resolve symbol __aeabi_memclr failed
```

after adding __aeabi_memclr to aot_reloc_arm.c (i have no idea if it's a correct thing to do or not):
```
1f034a3013ec# ./iwasm ../../../../wamr-compiler/b2/aot
Iteration number 0
(hanged up for some reasons)
```

yamt, 2023-08-14:
> (hanged up for some reasons)

the wasm file (the input of wamrc) did hang up in the same way. so i guess it isn't an aot issue.

Zzzabiyaka, 2023-08-14:
hm 

I will check the example, I was deleting some parts of it to make minimal example for another issue. 

Maybe I've provided you with the wrong one

Zzzabiyaka, 2023-08-14:
I'm building iwasm using `test_wamr.sh` (though it shouldn't be different) 

the following config is set 

```
 -DWAMR_BUILD_TARGET=X86_64 -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_FAST_INTERP=1 -DWAMR_BUILD_JIT=0 -DWAMR_BUILD_AOT=0 -DWAMR_BUILD_SPEC_TEST=1 -DCOLLECT_CODE_COVERAGE=0 -DWAMR_BUILD_BULK_MEMORY=1 -DWAMR_BUILD_MULTI_MODULE=0 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=0 -DWAMR_BUILD_LIB_WASI_THREADS=1
```

yamt, 2023-08-14:
>  I'm using clang from wasi-sdk 20 release by the way

i was using the same version.

off topic: the libc in wasi-sdk-20 has a few known bugs which can affect workloads like your test code.

yamt, 2023-08-14:
> I'm building iwasm using `test_wamr.sh` (though it shouldn't be different)
>
> the following config is set
>
> ```
>  -DWAMR_BUILD_TARGET=X86_64 -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_FAST_INTERP=1 [...]

sorry, aren't you using armv7?

Zzzabiyaka, 2023-08-14:
>off topic: the libc in wasi-sdk-20 has a few known bugs which can affect workloads like your test code.
yeah, we're aware of it

yamt, 2023-08-14:
> > > I'm building iwasm using `test_wamr.sh` (though it shouldn't be different)
> > > the following config is set
> > > ```
> > >  -DWAMR_BUILD_TARGET=X86_64 -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_FAST_INTERP=1 -DWAMR_BUILD_JIT=0 -DWAMR_BUILD_AOT=0 -DWAMR_BUILD_SPEC_TEST=1 -DCOLLECT_CODE_COVERAGE=0 -DWAMR_BUILD_BULK_MEMORY=1 -DWAMR_BUILD_MULTI_MODULE=0 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=0 -DWAMR_BUILD_LIB_WASI_THREADS=1
> > > ```
> > 
> > 
> > sorry, aren't you using armv7?
> 
> sorry, I've just pasted the config that I use on linux
> 
> Can you confirm that it works on linux for you?

i usually use macOS/x86-64. it seems working there, with or without aot. (until it hits libc bug)

> 
> Because for me it works on linux for both wasm and aot, and for arm works only .wasm

i want to reproduce the "resolve symbol .aot_stack_sizes failed" issue.

i might or might not investigate the hangup later. but it isn't my priority right now.

Zzzabiyaka, 2023-08-14:
Yeah I've double-checked 

I'm sorry, the issue that our wamr is a bit old (v1.2.2) but files were compiled with modern wamrc and obviously it didn't work 

I'll close the issue, sorry again

closed 2023-08-15 by Zzzabiyaka