wasm-micro-runtime/wasm-micro-runtime #2460: module instance context api

yamt, 2023-08-14:
## motivation

some native functions (eg. wasi) need to maintain some execution context. (eg. wasi file table)
for the in-tree code, the relevant logic can be hardcoded in the runtime. (eg. wasm_runtime_set_wasi_ctx, wasm_runtime_destroy_wasi)
however, it would be nice to have some api to maintain such a state a bit more dynamically because:
* for better modularity
* allow native functions maintained out-of-tree

## proposed api

```C
/*
 * module instance context APIs
 *   wasm_runtime_module_instance_context_key_create
 *   wasm_runtime_module_instance_context_key_destroy
 *   wasm_runtime_module_instance_set_context
 *   wasm_runtime_module_instance_set_context_spread
 *   wasm_runtime_module_instance_get_context
 *
 * This set of APIs is intended to be used by an embedder which provides
 * extra sets of native functions, which need per module instance state
 * and are maintained outside of the WAMR tree.
 *
 * It's modelled after the pthread specific API.
 *
 * wasm_runtime_module_instance_set_context_spread is similar to
 * wasm_runtime_module_instance_set_context, except that
 * wasm_runtime_module_instance_set_context_spread applies the change
 * to all threads in the cluster.
 * It's an undefined behavior if multiple threads in a cluster call
 * wasm_runtime_module_instance_set_context_spread on the same key
 * simultaneously. It's a caller's resposibility to perform necessary
 * serialization if necessary. For example:
 *
 * if (wasm_runtime_module_instance_get_context(inst, key) == NULL) {
 *     newctx = alloc_and_init(...);
 *     lock(some_lock);
 *     if (wasm_runtime_module_instance_get_context(inst, key) == NULL) {
 *         // this thread won the race
 *         wasm_runtime_module_instance_set_context_spread(inst, key, newctx);
 *         newctx = NULL;
 *     }
 *     unlock(some_lock);
 *     if (newctx != NULL) {
 *         // this thread lost the race, free it
 *         cleanup_and_free(newctx);
 *     }
 * }
 *
 * Note: dynamic key create/destroy while instances are live is not
 * implemented as of writing this.
 * it's caller's resposibility to ensure destorying all module instances
 * before calling wasm_runtime_module_instance_context_key_create or
 * wasm_runtime_module_instance_context_key_destroy.
 * otherwise, it's an undefined behavior.
 *
 * Note about threads:
 * - When spawning a thread, the contexts (the pointers given to
 *   wasm_runtime_module_instance_set_context) are copied from the parent
 *   instance.
 * - The destructor is called only on the main instance.
 */

WASM_RUNTIME_API_EXTERN void *
wasm_runtime_module_instance_context_key_create(
    void (*dtor)(wasm_module_inst_t inst, void *ctx));

WASM_RUNTIME_API_EXTERN void
wasm_runtime_module_instance_context_key_destroy(void *key);

WASM_RUNTIME_API_EXTERN void
wasm_runtime_module_instance_set_context(wasm_module_inst_t inst, void *key,
                                         void *ctx);
WASM_RUNTIME_API_EXTERN void
wasm_runtime_module_instance_set_context_spread(wasm_module_inst_t inst, void *key,
                                         void *ctx);
WASM_RUNTIME_API_EXTERN void *
wasm_runtime_module_instance_get_context(wasm_module_inst_t inst, void *key);
```

## proposed implementation

https://github.com/bytecodealliance/wasm-micro-runtime/pull/2436

## Alternative 1

make the embedder to maintain the state. it probably works well if the embedder and native functions are provided by the same group.

## Alternative 2

use `wasm_runtime_set_custom_data`.
it would work if only single state is necessary on the system.

it's easy to re-implement `wasm_runtime_set_custom_data` on the top of the proposed api.
the opposite is difficult.

## Alternative 3

(ab?)use externref api with https://github.com/bytecodealliance/wasm-micro-runtime/pull/2455

tonibofarull, 2023-08-28:
@lum1n0us thanks for the comment!
I don't want to get ahead of @yamt about the explanation, but my point of view is the following:

Although we can implement WASI libraries and implement context handling in WAMR, I think it is better to let these extensions handle the context by themselves without the need to modify the code.

For example,

https://github.com/bytecodealliance/wasm-micro-runtime/blob/ff151fb7bac109636a944149e545317c4b3d798a/core/iwasm/interpreter/wasm_runtime.c#L2239-L2241

This could be handled via the proposed API. This way, without needing to modify the WAMR's core code, we could enable wasi-nn through flags at build time or as a native library.

yamt, 2023-11-22:
closed by https://github.com/bytecodealliance/wasm-micro-runtime/pull/2436

closed 2023-11-22 by yamt