wasm-micro-runtime/wasm-micro-runtime #2469: aot doesn't seem to handle max pages correctly

yamt, 2023-08-15:
the following code traps. (x86-64)
```
(module
  (func (export "_start")
    i32.const 0
    i32.load offset=0xffff_fffc
    i32.const 0x1234_5678
    i32.ne
    if
      unreachable
    end
  )
  (memory (export "memory") 65536)
  (data (i32.const 0xffff_fffc) "\78\56\34\12")
)
```

wenyongh, 2023-08-16:
Hi, this is a defect currently, since the type of field memory_data_size in WASMMemoryInstance is uint32, its max value is UINT32_MAX and we don't want to change the AOT ABI currently, we resize the linear memory to 4G-1 if the initial memory count is 65536. It may take time and effort to fully resolve the issue, e.g. change memory_data_size't type to uint64 and change some related code. How about we release 1.2.3 firstly and fix the issue after that? Or do you suggest to fix the issue now? Thanks.

yamt, 2023-08-16:
i don't think this is a release blocker or anything too important. i filed this issue just for a reminder.

wenyongh, 2024-03-15:
The issue should have been resolved in PR https://github.com/bytecodealliance/wasm-micro-runtime/pull/3209, after changing the the field `memory_data_size` of `WASMMemoryInstance` from type `uint32` to `uint64`, and removing the related code to resize the memory data size from 4GB to UINT32.