wasm-micro-runtime/wasm-micro-runtime #2480: Cannot create module with `wasm_module_new` from `wasm_c_api.h`

schorg, 2023-08-18:
When using the Wasm C API (`wasm_c_api`) I always get the same error when I try to call

```
wasm_module_t* = wasm_module_new(store, &binary);
...
[12:36:58:309 - FFFF54CCAE90]: WASM module load failed: unexpected end of section or function
```

I tried this with several - extremely simple - Wasm modules and always get the same error. 
It also does not load simple examples (`hello.wasm`) from the Wasm C API examples.

I use the latest WASI-SDK for creating the Wasm modules, and the latest WAMR from the `main` branch.

Here is a complete log. `INFO` and `ERROR` are my own logs.  The rest comes from the wasm_c_api function calls.
By the way, the same example executes without errors with `iwasm dummy.wasm`.

```
2023-08-18 10:37:17,622 INFO [default] Loading binary: dummy.wasm
2023-08-18 10:37:17,623 INFO [default] Validating module ...
[12:36:58:869 - FFFF54CCAE90]: Load type section success.

[12:36:58:119 - FFFF54CCAE90]: Load import section success.

[12:36:58:142 - FFFF54CCAE90]: Load function section success.

[12:36:58:147 - FFFF54CCAE90]: Load table section success.

[12:36:58:151 - FFFF54CCAE90]: Load memory section success.

[12:36:58:158 - FFFF54CCAE90]: Load global section success.

[12:36:58:163 - FFFF54CCAE90]: Load export section success.

[12:36:58:166 - FFFF54CCAE90]: Load code segment section success.

[12:36:58:232 - FFFF54CCAE90]: /home/ubuntu/Projects/rds/ctrlx-wasm-rt/third-party/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c, line 714, can not find an export 0 named _initialize in the module
[12:36:58:245 - FFFF54CCAE90]: Load module success.

2023-08-18 10:37:17,625 INFO [default] Compiling module...
[12:36:58:303 - FFFF54CCAE90]: Load type section success.

[12:36:58:309 - FFFF54CCAE90]: WASM module load failed: unexpected end of section or function
[12:36:58:322 - FFFF54CCAE90]: wasm_module_new failed
2023-08-18 10:37:17,625 ERROR [default] Error compiling module: dummy.wasm
```
Thanks, for any help.

schorg, 2023-08-18:
Some insights from debugging.

If I call `wasm_module_validate(store, binary);` first and `wasm_module_new(store, binary)` second, I get this error.

If I reverse the order and call `wasm_module_new` first and `wasm_module_validate` second, I do not see the error.

Inspecting the code shows that `wasm_module_new` creates a copy of the `binary`, while `wasm_module_validate` doesn't. Both functions call `wasm_runtime_load` with the binary. 

I suspect, `wasm_runtime_load` somehow changes the binary.  

I have not enough insight into the implementation to understand where this happens.

lum1n0us, 2023-08-21:
Thanks for catching it. It seems at the end of `wasm_module_validate()`, `wasm_unload()` will release the binary content.

lum1n0us, 2023-08-21:
#2483

schorg, 2023-08-22:
Thanks for fixing this. Now it works as expected.

closed 2023-08-22 by schorg