wasm-micro-runtime/wasm-micro-runtime #2495: Proposed implementation of WASI libc filesystem on Windows

zoraaver, 2023-08-22:
## Motivation

We would like to have a working implementation of all the WASI libc filesystem functions on Windows. This is currently not possible due to the use of various [POSIX file functions](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c) which are not available on Windows or are not fully [POSIX compliant](https://learn.microsoft.com/en-us/cpp/c-runtime-library/compatibility?view=msvc-170). 

## Requirements

The solution outlined in this proposal is not necessarily specific to POSIX filesystem functions (it could be equally applied to the POSIX time functions for example) but we will only concern ourselves with filesystem functions. The solution can be easily extended to other POSIX functions in the future.

## Proposal

The core issue is that the WASI libc implementation assumes the underlying platform is a POSIX platform. In the codebase, there is already an existing method of wrapping platform-specific APIs: a ‘wrapper’ function is defined in an interface header file and then implemented on a per-platform basis. e.g. `os_mutex_lock` is declared in [core/shared/platform/include/platform_api_vmcore.h](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/include/platform_api_vmcore.h#L97) and is implemented using platform-specific APIs for each platform under [core/shared/platform](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/core/shared/platform) (see [Windows implementation](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/windows/win_thread.c#L513) and [POSIX implementation](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/common/posix/posix_thread.c#L144)). This approach has the advantage of avoiding the use of lots of `#ifdef`’s when we need to use any platform-specific APIs. The proposal here is to adopt the same approach for the POSIX filesystem functions: wrap them in platform-agnostic interface functions which are implemented on a per platform basis. This will mean the WASI libc implementation will no longer reference any POSIX file functions directly, but rather wrapper functions which will call the underlying POSIX function on POSIX platforms and call the equivalent Windows APIs on Windows. Obviously we are only interested in Windows here, but this solution applies to any non-POSIX compliant platform. The benefit of this solution is that it allows us to keep the majority of the WASI libc implementation the same since most of the code is platform-agnostic already.

This approach is very similar to that taken by [WasmEdge](https://github.com/WasmEdge); there are both Windows and Linux classes to wrap any platform-specific details: [Windows](https://github.com/WasmEdge/WasmEdge/blob/master/lib/host/wasi/inode-win.cpp), [Linux](https://github.com/WasmEdge/WasmEdge/blob/master/lib/host/wasi/inode-linux.cpp). The implementation is in C++ but it will still be useful to use as a reference when implementing POSIX file functionality for Windows. (the Linux implementation mostly passes through parameters to the underlying POSIX functions as expected).

## Implementation

#### Wrapper Function Naming

To make it easy to understand what a wrapper function does without digging into a specific implementation, we can name them according to their equivalent POSIX function but with an addition `os_` prefix. e.g. `preadv` would become `os_preadv`.

#### Abstracting Platform Handles

Aside from relying on the existence of various POSIX functions/data structures, we also assume that a platform handle (a.k.a. file descriptor on POSIX) is of an `int`  type [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c#L423) which will not be true in general. In order to support Windows and any other non-POSIX platforms, we will need to define this type on a per platform basis.

Instead of `int`, we can use a type `os_handle` , which would be typedef’ed to `int` on POSIX. On Windows, we would define it as follows:

```
typedef struct windows_handle {
  HANDLE handle;
  __wasi_fdflags_t fdflags;
} os_handle;
```

 We need to store the fd flags separately from the raw Win `HANDLE` since there is no way of retrieving/storing these flags on a Windows `HANDLE` natively.

The POSIX wrapper functions would then accept/return this `os_handle` type where appropriate.

#### Wrapper Function Signature

We can keep the signature of each wrapper function the same as the equivalent POSIX function, with the exception of the following changes:

* Explicitly return a WASI error (`__wasi_errno_t`) from each function rather than using `errno`. `errno` is not set by Windows APIs and while we could set it ourselves, it is not a very clean way to handle errors and its usage as far as possible should be restricted to the POSIX implementation. Where the original POSIX function returned a value (that is not already an error code), we can instead use an additional out parameter. Each platform will then be responsible for converting their native error code into the appropriate WASI error code.
* Where the equivalent POSIX function would expect a data structure/type defined in the POSIX specification, we would instead use the corresponding WASI data structure, which is necessarily platform-agnostic. e.g.
    * `ssize_t writev(int *fildes*, const struct iovec *`*`iov`*`, int `*`iovcnt`*`);` would be wrapped in the following function:  `__wasi_errno_t os_writev(os_handle *fildes*, const struct __wasi_iovec_t**iov*, int *iovcnt*, ** ssize_t **bytes_written*);`

This would include POSIX flags e.g.  `O_DIRECTORY`, `O_APPEND`. For all the functions we need to implement, there are corresponding WASI flags e.g. `__WASI_O_DIRECTORY` , `__WASI_FDFLAG_APPEND`.

#### Wrapper Function Declaration

The most suitable place to declare these functions seems to be [core/shared/platform/include/platform_api_extension](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/include/platform_api_extension.h)[.h](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/include/platform_api_extension.h).

#### Standard I/O Streams

In addition to wrapping the standard POSIX filesystem functions, we will need a way to obtain handles to `stdin`, `stdout` and `stderr` (if they are available at all on the platform). Currently, we default these to the values [0, 1 and 2](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_runtime_common.c#L2884-L2887) respectively which would only be correct on POSIX systems. Therefore, we can introduce an additional wrapper function

* `os_handle os_get_std_handle(os_std_handle std_handle);`

together with constants to represent the stdin/stdout/stderr devices:

* `OS_STD_OUTPUT_HANDLE`
* `OS_STD_INPUT_HANDLE`
* `OS_STD_ERROR_HANDLE`

defining them as appropriate on POSIX and [Windows](https://learn.microsoft.com/en-us/windows/console/getstdhandle).

N.B. on POSIX, `os_get_std_handle` would just pass through the std handle without modification.

#### Directory Handles

We rely on the existence of POSIX directory handles (`DIR*`) [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c#L429). Similar to [Abstracting Platform Handles](#abstracting-platform-handles), we will need to define this directory handle type as `os_directory_handle` on a per-platform basis. On POSIX, it would be defined to `DIR` whereas on Windows, we will need to emulate `DIR`. We can use this [port](https://github.com/win32ports/dirent_h/blob/master/dirent.h#L116) of `dirent.h` to Windows as a reference for the implementation of the specific structs and directory functions.
[... 101 lines cut ...]
|fd_sync	|N	|I	|C	|
|fd_tell	|N	|E	|C	|
|fd_write	|E	|E	|C	|
|path_create_directory	|I	|I	|C	|
|path_filestat_get	|I	|E	|C	|
|path_filestat_set_times	|N	|E	|C	|
|path_link	|N	|E	|C	|
|path_open	|I	|E	|C	|
|path_readlink	|N	|E	|C	|
|path_remove_directory	|I	|I	|C	|
|path_rename	|N	|E	|C	|
|path_symlink	|I	|E	|C	|
|path_unlink_file	|I	|I	|C	|

Test status:

* I = indirectly tested i.e. function is used in other tests but there is no explicit test for that function
* E = explicitly tested, there are one or more tests dedicated to testing that function
* N = not tested at all

Action:

* C = copy relevant tests from wasmtime to WASI test suite
* W = write new test to fill testing gap
* D = testing is sufficient (whether indirect/explicit) so no action is required

yamt, 2023-09-26:
> The issue with this approach is that all of these WASI functions need to [look up the host handle from the WASI [...]

it's necessary for other approaches too, isn't it?

zoraaver, 2023-10-12:
> it's necessary for other approaches too, isn't it?

Yes, we always have to look up the host handle from the WASI fd, regardless of the approach. But if we implement the WASI functions for each platform, this logic would be duplicated for each platform since it is platform-agnostic.

zoraaver, 2024-02-29:
All the work has been merged to main. The only WASI function that remains to be implemented on Windows is `poll_oneoff`.