wasm-micro-runtime/wasm-micro-runtime #2513: posix shared memory with wamr

VaibhavVgithub, 2023-08-29:
[01:25:19:899 - 7F01820E5580]: warning: failed to link import function (env, shm_open)
[01:25:19:909 - 7F01820E5580]: warning: failed to link import function (env, ftruncate)
Exception: failed to call unlinked import function (env, shm_open)

----------
can someone please explain how to resolve this issue. I want shared memory functionalities in my 'main.c'. And also, Can someone suggest any other procedure than [native-lib](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/native-lib) example to use shared memory applications with wamr?

lum1n0us, 2023-08-30:
May I ask:
- which toolchain, (like wasi-sdk, emscripten, or something else), is used to generate the .wasm?
- what kind of compilation option groups is used to generate the .wasm?

VaibhavVgithub, 2023-08-30:
I am using wasi-sdk 20.0 to generate .wasm

------
set (CMAKE_SYSTEM_PROCESSOR wasm32)
#set (CMAKE_SYSROOT                  ${WAMR_ROOT_DIR}/wamr-sdk/app/libc-builtin-sysroot)

if (NOT DEFINED WASI_SDK_DIR)
    set (WASI_SDK_DIR               "/opt/wasi-sdk")
endif ()

set (CMAKE_C_FLAGS                  "-nostdlib -pthread -Qunused-arguments -D_WASI_EMULATED_MMAN")
set (CMAKE_C_COMPILER_TARGET        "wasm32-wasi")
set (CMAKE_C_COMPILER               "${WASI_SDK_DIR}/bin/clang")

set (CMAKE_EXE_LINKER_FLAGS
    "-Wl,--max-memory=131072, -z stack-size=32768   \
     -Wl,--shared-memory \
     -g \
     -lwasi-emulated-mman \
     -Wl,--export-all \
     -Wl,--no-entry,--strip-all                   \
     -Wl,--export=__main_argc_argv                \
     -Wl,--export=__heap_base, -Wl,--export=__data_end \
     -Wl,--allow-undefined \
     -pthread"
)
add_executable(test.wasm main.c)
target_link_libraries(test.wasm)
--------------
this is what I am using to create "test.wasm"

lum1n0us, 2023-08-31:
`-nostdlib` might be the reason. Since `shm_open` and `ftruncate` are all provided by WASI-SDK.

VaibhavVgithub, 2023-08-31:
Hi, No, That can't be reason atleast, because I earlier didn't use "-nostdlib" but that gave me below error, 
-------------------

In file included from /home/vaibhav/Desktop/wasm-micro-runtime-main/samples/shm_open/wasm-app/main.c:8:
/opt/wasi-sdk-20.0/bin/../share/wasi-sysroot/include/sys/mman.h:2:2: error: "WASI lacks a true mmap; to enable minimal mmap emulation, compile with -D_WASI_EMULATED_MMAN and link with -lwasi-emulated-mman"
#error "WASI lacks a true mmap; to enable minimal mmap emulation, \
 ^
/home/vaibhav/Desktop/wasm-micro-runtime-main/samples/shm_open/wasm-app/main.c:78:20: error: call to undeclared function 'shm_open'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
    int shmem_fd = shm_open(mem, oflag, mode);
                   ^
2 errors generated.
make[2]: *** [wasm-app/CMakeFiles/test.wasm.dir/build.make:76: wasm-app/CMakeFiles/test.wasm.dir/main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:210: wasm-app/CMakeFiles/test.wasm.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

-------------------
and like this
---------------

[100%] Linking C executable test.wasm
wasm-ld: error: --shared-memory is disallowed by dlmalloc.o because it was not compiled with 'atomics' or 'bulk-memory' features.
wasm-ld: error: 'atomics' feature must be used in order to use shared memory
clang-16: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [wasm-app/CMakeFiles/test.wasm.dir/build.make:97: wasm-app/test.wasm] Error 1
make[1]: *** [CMakeFiles/Makefile2:210: wasm-app/CMakeFiles/test.wasm.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

---------------------

so I added those options after taking reference from wamr docs

lum1n0us, 2023-09-03:
😵‍💫 My mistake. WASI has limited support on mmap (https://github.com/WebAssembly/wasi-libc/blob/main/libc-bottom-half/mman/mman.c) and is lack of IPC supporting. There is ongoing discussion about [IPC module](https://github.com/WebAssembly/wasi-libc/blob/main/libc-bottom-half/mman/mman.c). 

On the opposite, EMSDK has shared memory supporting but lack of file APIs unless trying [WASMFS](https://github.com/emscripten-core/emscripten/blob/main/src/settings.js#L1735C5-L1735C11)