wasm-micro-runtime/wasm-micro-runtime #2547: Integer too large: running iwasm on aarch64 linux

hardiksd, 2023-09-13:
Hi,

I'm trying to run WASM WAMR on linux aarch64 openwrt. 

I have successfully compiled linux library and iwasm for my platform. But when I try to run a wasm module I see following error.

root@OpenWrt:~# ./iwasm --addr-pool=0.0.0.0/15 tcp_server.wasm
WASM module load failed: integer too large

I'm trying to make sense of it. I have double checked that my iwasm is 64 bit and Aarch64. This error above seems to indicate some issue with endianess. Any pointers? For samples do I need to instruct the compiler to cross compile? I ran the .wasm file on x64 machine without any issues.

wenyongh, 2023-09-13:
Hi, could you try building iwasm with multi-thread support? `cmake -DWAMR_BUILD_LIB_PTHREAD=1`.

hardiksd, 2023-09-13:
Thanks. that was it. I was using iwasm compiled for minimal example and it didn't have pthread enabled. Appreciate the prompt reply.

closed 2023-09-13 by hardiksd