wasm-micro-runtime/wasm-micro-runtime #2555: The results of execution for a wasm binary are different from other wasm runtimes.

abc767234318, 2023-09-17:
I constructed a wasm binary with an exported function called "main". I used the following command and compared the outputs of WAMR with other runtimes.
[file3.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12642325/file3.zip)

```
wamrc -o file3.wasm.aot file3.wasm
iwasm --heap-size=0  -f main file3.wasm.aot
```
The outputs of WAMR are as follows:
```
Create AoT compiler with:
  target:        x86_64
  target cpu:    znver3
  target triple: x86_64-unknown-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file file3.wasm.aot was generated.
Exception: unreachable
```

While the outputs of the other runtimes like wasmtime and wasmer are:
```
-----------------wasmtime-----------------
Error: failed to run main module `file3.wasm`

Caused by:
    0: failed to invoke `main`
    1: error while executing at wasm backtrace:
           0:  0x741 - <unknown>!<wasm function 5>
           1:  0x7c6 - <unknown>!<wasm function 6>
           2: 0x16fd - <unknown>!<wasm function 16>
    2: wasm trap: out of bounds memory access

-----------------wasmer-----------------
error: failed to run `file3.wasm`
╰─▶ 1: RuntimeError: out of bounds memory access
           at <unnamed> (file3.wasm[5]:0x741)
           at <unnamed> (file3.wasm[6]:0x7c6)
           at <unnamed> (file3.wasm[16]:0x16fd)
```

The wasm binary should have an out-of-bounds memory access error, but WAMR encountered an unreachable instruction.

yamt, 2023-09-20:
how have you generated these files? just curious.

abc767234318, 2023-09-20:
> how have you generated these files? just curious.

I wrote a fuzzer that would generate syntactically and formationally compliant wasm binary.

lum1n0us, 2023-09-28:
Because WAMR disables bounds check by default in .aot files. If using bounds check likes: `wamrc --bounds-checks=1 -o file3.aot file3.wasm`, a similar error message, *Exception: out of bounds memory access*, will be found.

lum1n0us, 2023-09-28:
@yamt if using llvm libraries with assertions. `wamrc --bounds-checks=1 -o files3.aot file3.wasm` will output an assertion failure:

```
wamrc: /workspaces/wasm-micro-runtime/core/deps/llvm/llvm/include/llvm/Object/ObjectFile.h:310: uint64_t llvm::object::ObjectFile::getCommonSymbolSize(llvm::object::DataRefImpl) const: Assertion `*SymbolFlagsOrErr & SymbolRef::SF_Common' failed.
```

its call stack likes

```
__GI___assert_fail (@7ffff74c4e32..7ffff74c4ebb:3)
llvm::object::ObjectFile::getCommonSymbolSize(llvm::object::DataRefImpl) const (\workspaces\wasm-micro-runtime\core\deps\llvm\llvm\include\llvm\Object\ObjectFile.h:310)
aot_resolve_stack_sizes (\workspaces\wasm-micro-runtime\core\iwasm\compilation\aot_emit_aot_file.c:2706)
aot_resolve_functions (\workspaces\wasm-micro-runtime\core\iwasm\compilation\aot_emit_aot_file.c:2851)
aot_obj_data_create (\workspaces\wasm-micro-runtime\core\iwasm\compilation\aot_emit_aot_file.c:3567)
aot_emit_aot_file_buf (\workspaces\wasm-micro-runtime\core\iwasm\compilation\aot_emit_aot_file.c:3582)
aot_emit_aot_file (\workspaces\wasm-micro-runtime\core\iwasm\compilation\aot_emit_aot_file.c:3650)
main (\workspaces\wasm-micro-runtime\wamr-compiler\main.c:663)
```

triggered by `uint64 sz = LLVMGetSymbolSize(sym_itr);`

yamt, 2023-09-28:
> @yamt if using llvm libraries with assertions. `wamrc --bounds-checks=1 -o files3.aot file3.wasm` will output an assertion failure:
>
> ```
> wamrc: [...]

i'll take a look.

yamt, 2023-09-29:
> i'll take a look.

https://github.com/bytecodealliance/wasm-micro-runtime/pull/2608

lum1n0us, 2023-11-16:
although `--bound-checks=1` are helpful, there are still concerns.

https://github.com/bytecodealliance/wasm-micro-runtime/issues/2773 explains the root cause and options, please let us know your opinion.