wasm-micro-runtime/wasm-micro-runtime #2556: WAMR has a segmentation fault while executing this wasm binary.

abc767234318, 2023-09-17:
I have generated a wasm binary, and I process it with Wamr. But the output error message differs from other runtimes like wasmtime and wasmer.
[file7.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12642331/file7.zip)
```
wamrc -o file7.wasm.aot file7.wasm
iwasm --heap-size=0  -f main file7.wasm.aot
```
The outputs of WAMR are as follows:
```
-----------------wamr-----------------
Create AoT compiler with:
  target:        x86_64
  target cpu:    znver3
  target triple: x86_64-unknown-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file file7.wasm.aot was generated.
Segmentation fault (core dumped)
```
While the outputs of the other runtimes like wasmtime and wasmer are:
```
-----------------wasmtime-----------------
warning: using `--invoke` with a function that returns values is experimental and may break in the future
0
0
0
0
0
0
0
520359494

-----------------wasmer-----------------
0 0 0 0 0 0 0 520359494
```

TianlongLiang, 2023-09-22:
Hi, the pr #2583 should fix this issue. As for the other issues you mentioned, we are actively debugging them to see where they go wrong. 

BTW, kudos to the fuzzer you programmed! It is really great that you spotted those issues all at once.

wenyongh, 2024-01-11:
Close this issue as it had been fixed.

closed 2024-01-11 by wenyongh