wasm-micro-runtime/wasm-micro-runtime #2559: The aot compiler encountered a segmentation fault while compiling this wasm binary.

abc767234318, 2023-09-17:
I constructed a wasm binary with an exported function called "main". I used the following command to compile it with aot compiler, while a segmentation fault was encountered.
[file30.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12642865/file30.zip)

```
wamrc -o file30.wasm.aot file30.wasm
```

And the outputs are:
```
Segmentation fault (core dumped)
```

While the outputs of this binary in the other runtimes like wasmtime and wasmer are:
```
-----------------wasmtime-----------------
warning: using `--invoke` with a function that returns values is experimental and may break in the future
79228157791897854723898736640
79228157791897854723898736640
79228157791897854723898736640
79228157791897854723898736640
79228157791897854723898736640
79228157791897854723898736640
79228157791897854723898736640

-----------------wasmer-----------------
79228157791897854723898736640 79228157791897854723898736640 79228157791897854723898736640 79228157791897854723898736640 79228157791897854723898736640 79228157791897854723898736640 79228157791897854723898736640
```

lum1n0us, 2023-09-26:
#2595

wenyongh, 2024-01-11:
Close this issue as it has been fixed.

closed 2024-01-11 by wenyongh