wasm-micro-runtime/wasm-micro-runtime #2561: The results of execution for a wasm binary are different from other wasm runtimes.

abc767234318, 2023-09-17:
I constructed a wasm binary with an exported function called "main". I used the following command and compared the outputs of WAMR with other runtimes.
[file76.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12642938/file76.zip)

```
wamrc -o file76.wasm.aot file76.wasm
iwasm --heap-size=0  -f main file76.wasm.aot
```
The outputs of WAMR are as follows:
```
Create AoT compiler with:
  target:        x86_64
  target cpu:    znver3
  target triple: x86_64-unknown-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file file76.wasm.aot was generated.
<0x00004febf1445b2b 0x0000000000000000>:v128,<0x00004febf1445b2b 0x0000000000000000>:v128,<0x00004febf1445b2b 0x0000000000000000>:v128,<0x00004febf1445b2b 0x0000000000000000>:v128,<0x00004febf1445b2b 0x0000000000000000>:v128
```

While the outputs of the other runtimes like wasmtime and wasmer are:
```
-----------------wasmtime-----------------
warning: using `--invoke` with a function that returns values is experimental and may break in the future
1099511585569
1099511585569
1099511585569
1099511585569
1099511585569

-----------------wasmer-----------------
1099511585569 1099511585569 1099511585569 1099511585569 1099511585569
```

lum1n0us, 2023-09-28:
#2584

wenyongh, 2024-01-11:
Close this issue since it has been fixed.

closed 2024-01-11 by wenyongh