wasm-micro-runtime/wasm-micro-runtime #2562: There may be a boundary check error in the store or load instruction of simd in WAMR.

abc767234318, 2023-09-17:
I constructed a wasm binary with an exported function called "main". I used the following command and compared the outputs of WAMR with other runtimes.
[file10.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12642994/file10.zip)

```
wamrc -o file10.wasm.aot file10.wasm
iwasm --heap-size=0  -f main file10.wasm.aot
```

The outputs of WAMR are as follows:
```
-----------------wamr-----------------
Create AoT compiler with:
  target:        x86_64
  target cpu:    znver3
  target triple: x86_64-unknown-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file file10.wasm.aot was generated.
<0xffffffffffffffff 0x00000000ffff0000>:v128,<0xffffffffffffffff 0x00000000ffff0000>:v128,<0xffffffffffffffff 0x00000000ffff0000>:v128,<0xffffffffffffffff 0x00000000ffff0000>:v128,<0xffffffffffffffff 0x00000000ffff0000>:v128,<0xffffffffffffffff 0x00000000ffff0000>:v128,<0xffffffffffffffff 0x00000000ffff0000>:v128
```

While the outputs of the other runtimes like wasmtime and wasmer are:
```
file10.wasm
-----------------wasmtime-----------------
Error: failed to run main module `file10.wasm`

Caused by:
    0: failed to invoke `main`
    1: error while executing at wasm backtrace:
           0:  0xc47 - <unknown>!<wasm function 10>
    2: wasm trap: out of bounds memory access

-----------------wasmer-----------------
error: failed to run `file10.wasm`
╰─▶ 1: RuntimeError: out of bounds memory access
           at <unnamed> (file10.wasm[10]:0xc47)

-----------------wasmedge-----------------
[2023-09-17 21:45:00.554] [error] execution failed: out of bounds memory access, Code: 0x88
[2023-09-17 21:45:00.554] [error]     Accessing offset from: 0x100000088 to: 0x10000008b , Out of boundary: 0xffffffff
[2023-09-17 21:45:00.554] [error]     In instruction: v128.load32_zero (0xfd 0x5c) , Bytecode offset: 0x00000c47
[2023-09-17 21:45:00.554] [error]     When executing function name: "main"
```

lum1n0us, 2023-09-28:
refer to https://github.com/bytecodealliance/wasm-micro-runtime/issues/2555#issuecomment-1738527008

lum1n0us, 2023-11-16:
although `--bound-checks=1` are helpful, there are still concerns.

https://github.com/bytecodealliance/wasm-micro-runtime/issues/2773 explains the root cause and options, please let us know your opinion.