wasm-micro-runtime/wasm-micro-runtime #2563: WAMR encountered an unexpected endless loop while executing a wasm binary.

abc767234318, 2023-09-17:
I constructed a wasm binary with an exported function called "main". I used the following command and compared the outputs of WAMR with other runtimes.

[file16.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12643019/file16.zip)

```
wamrc -o file16.wasm.aot file16.wasm
iwasm --heap-size=0  -f main file16.wasm.aot
```

WAMR has no output and appears to be in an endless loop. While the outputs of the other runtimes like wasmtime and wasmer are:

```
-----------------wasmtime-----------------
Error: failed to run main module `file16.wasm`

Caused by:
    0: failed to invoke `main`
    1: error while executing at wasm backtrace:
           0:  0xbd1 - <unknown>!<wasm function 5>
           1: 0x61eb - <unknown>!<wasm function 45>
    2: wasm trap: out of bounds memory access

-----------------wasmer-----------------
error: failed to run `file16.wasm`
╰─▶ 1: RuntimeError: out of bounds memory access
           at <unnamed> (file16.wasm[5]:0xbd1)
           at <unnamed> (file16.wasm[45]:0x61eb)

-----------------wasmedge-----------------
[2023-09-17 21:47:12.839] [error] execution failed: out of bounds memory access, Code: 0x88
[2023-09-17 21:47:12.839] [error]     Accessing offset from: 0x100001fab to: 0x100001fb2 , Out of boundary: 0xffffffff
[2023-09-17 21:47:12.839] [error]     In instruction: v128.load8x8_u (0xfd 0x02) , Bytecode offset: 0x00000bd1
[2023-09-17 21:47:12.839] [error]     When executing function name: "main"
```

lum1n0us, 2023-09-28:
please refer to https://github.com/bytecodealliance/wasm-micro-runtime/issues/2555#issuecomment-1738527008

lum1n0us, 2023-11-16:
although `--bound-checks=1` are helpful, there are still concerns.

https://github.com/bytecodealliance/wasm-micro-runtime/issues/2773 explains the root cause and options, please let us know your opinion.