wasm-micro-runtime/wasm-micro-runtime #2564: The results of execution for a wasm binary are different from other wasm runtimes.

abc767234318, 2023-09-17:
I constructed a wasm binary with an exported function called "main". I used the following command and compared the outputs of WAMR with other runtimes.
[file65.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12643047/file65.zip)
```
wamrc -o file65.wasm.aot file65.wasm
iwasm --heap-size=0  -f main file65.wasm.aot
```

The outputs of WAMR are as follows:
```
-----------------wamr-----------------
Create AoT compiler with:
  target:        x86_64
  target cpu:    znver3
  target triple: x86_64-unknown-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file file65.wasm.aot was generated.
<0x0000000000001600 0x0000000000000000>:v128,<0x0000000000001600 0x0000000000000000>:v128,<0x0000000000001600 0x0000000000000000>:v128,<0x0000000000001600 0x0000000000000000>:v128,<0x0000000000001600 0x0000000000000000>:v128,<0x0000000000001600 0x0000000000000000>:v128,<0x0000000000001600 0x0000000000000000>:v128
```

While the outputs of the other runtimes like wasmtime and wasmer are:
```
-----------------wasmtime-----------------
warning: using `--invoke` with a function that returns values is experimental and may break in the future
4290778624
4290778624
4290778624
4290778624
4290778624
4290778624
4290778624

-----------------wasmer-----------------
4290778624 4290778624 4290778624 4290778624 4290778624 4290778624 4290778624
```

lum1n0us, 2023-09-20:
Is there a related source code file in any guest language? like C

abc767234318, 2023-09-20:
> Is there a related source code file in any guest language? like C

Without the related source code, this wasm binary is just randomly generated.

lum1n0us, 2023-10-16:
It is caused by variant implementation of `nan:arithmetic`. And I think it is **not a problem**.

It all starts from a `f32x4.min` operation with two operands, `v128 i32x4 0xffffffff 0x0 0x0 0x0` and `v128 i32x4 0x0 0x0 0x0 0x0`. Obviously, they actually are `v128 f32x4 nan 0.0 0.0 0.0` and `v128 f32x4 0.0 0.0 0.0 0.0`. According to Wasm spec, in this circumstance (nan vs 0.0), `f32x4.min` should return `nan`.

We all know, beside *expon*, *signif* of a `nan` can be variant, like (`nan("")`, `nan("1")`, `nan("2")`). In IEEE 754, it suggests returning `qnan` and `snan` based on if wants to trigger an exception or not. But in Wasm spec, there is no `qnan` and `snan`. instructions should return `nan` with arbitrary content. So, you will see, some runtimes return `0x7fc00000` and wamr returns `0xffffffff`. Both are NaN but with variant payload.

Further, following operations should deal NaN values as an exception, quit earlier or handle it specifically. Otherwise, it is no sense to keep arithmetic operations with NaN values.

https://webassembly.github.io/spec/core/syntax/values.html#floating-point
https://webassembly.github.io/spec/core/exec/numerics.html#nan-propagation

abc767234318, 2023-10-19:
> It is caused by variant implementation of `nan:arithmetic`. And I think it is **not a problem**.
>
> It all starts from a `f32x4.min` operation with two operands, `v128 i32x4 0xffffffff 0x0 0x0 0x0` [...]

Thank you very much for taking the time to deal with this issue. Although keeping arithmetic operations with NaN values has no meaning, I think wamr should be consistent with other runtimes. I compared four runtimes, wasmtime, wasmer, wamr and wasmedge, and I found that only wamr produced inconsistent results with the others.