wasm-micro-runtime/wasm-micro-runtime #2566: Did `WAMR-1.2.3` get re-tagged?

chenrui333, 2023-09-19:
👋 while regression build [wasm-micro-runtime for macOS sonoma](https://github.com/Homebrew/homebrew-core/actions/runs/6228113263/job/16904176720), we found that the checksum got change. Raise this issue to confirm if that is intentional. Thanks!

relates to https://github.com/Homebrew/homebrew-core/pull/143834

wenyongh, 2023-09-19:
Hi, yes, we re-tagged several times since there were some issues (e.g. compile errors) when running CIs to release the binaries, we had to fix these issues and ran the release process several times:
https://github.com/bytecodealliance/wasm-micro-runtime/actions/workflows/release_process.yml
https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/5933313825
https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/5935662965

Some PRs to fix the issues:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/2491
https://github.com/bytecodealliance/wasm-micro-runtime/pull/2492
https://github.com/bytecodealliance/wasm-micro-runtime/pull/2497

chenrui333, 2023-09-20:
Thanks @wenyongh for confirming it!

closed 2023-09-20 by chenrui333