wasm-micro-runtime/wasm-micro-runtime #2573: run jit enabled iwasm on aarch64 raspberrypi

y-ykcir, 2023-09-21:
Hi, I want to use JIT-mode iwasm on an aarch64 Raspberry Pi. 

First of all, I would like to ask if aarch64 already supports JIT-mode iwasm？
And Is it possible to cross-compile iwasm on x86 hosts or directly on Raspberry Pi?

I try to  use `CC=aarch64-linux-gnu-gcc cmake .. -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_PLATFORM=linux -DWAMR_BUILD_TARGET=AARCH64` but met error when  I `make`:

```
...
Relocations in generic ELF (EM: 183)
CMakeFiles/iwasm.dir/main.c.o: error adding symbols: File in wrong format
...
```

I would be grateful for any advice on iwasm running jit mode on the aarch64 Raspberry Pi, thanks!

liangzhupic, 2023-09-25:
same problem here. any solution for it?

y-ykcir, 2023-09-29:
I successfully compiled jit-mode iwasm directly on aarch64 Raspberry PI, but when I ran it, I got the following error:
```
pi@raspberrypi:~/WasmRL-PPO/wasm-rl-ppo/benchmark $ /home/pi/arm-wasm/iwasm --llvm-jit poly.wasm
Invalid thread stack size 65536. Min stack size on Linux = 131072
WASM module load failed: create orcjit compile thread failed

pi@raspberrypi:~/WasmRL-PPO/wasm-rl-ppo/benchmark $ /home/pi/arm-wasm/iwasm_jit --llvm-jit --stack-size=131072 poly.wasm
Invalid thread stack size 65536. Min stack size on Linux = 131072
WASM module load failed: create orcjit compile thread failed
```
@wenyongh Could you please take a look? Thanks a lot.

wenyongh, 2023-09-30:
@y-ykcir Could you add `add_definitions(-DAPP_THREAD_STACK_SIZE_DEFAULT=131072)` in the CMakeLists.txt?

y-ykcir, 2023-09-30:
> @y-ykcir Could you add `add_definitions(-DAPP_THREAD_STACK_SIZE_DEFAULT=131072)` and `add_definitions(-DAPP_THREAD_STACK_SIZE_MIN=131072)` in the CMakeLists.txt?

It worked. Thanks a lot. 😀

closed 2023-10-03 by y-ykcir