wasm-micro-runtime/wasm-micro-runtime #2587: WAMR IDE: Windows experience

svits-llc, 2023-09-25:
Hello, dears!

I do appreciate your effort on creation of vscode extension and am looking forward to see it thrive!
Let me be your tester for a while.

This is Windows 10, I am doing very basic project creation with "hello world" main.cpp

[image]

Compiles well, but can't run or debug due to:
```
[02:42:16:568 - 7F96F7C09140]: warning: allocate 28 bytes memory failed
Exception: allocate memory failed
```

Please let me know how can I help you to debug and fix this problem?

wenyongh, 2023-09-30:
Hi, could you try enlarging the `Max linear memory size`, and adding malloc and free in the `Exported symbols`?

svits-llc, 2023-10-04:
How? Can you show me example? This is a 3 line "hello world" application.

TianlongLiang, 2023-10-20:
Hi, you can modify the configuration directly, maybe try to double the `Max linear memory size`, and add malloc and free to `Exported symbols`, like `main, malloc, free`. After the modification, don't forget to hit "Modify" button
[image]

svits-llc, 2023-10-21:
[image]
It looks ever loading. And click on Configuration only causes this message:
[image]
But never opens the configuration

Is there a log I could send you to see details?

TianlongLiang, 2023-10-23:
Hi, are you using Win 11 or Win 10? I will try it on my windows and see if I encounter the same problem. If everything is fine at my end, I will tell you my steps and maybe you can follow them and see whether the problem persists.

svits-llc, 2023-10-23:
Win10. Thank you for checking this out for me! I appreciate it!

LevelCA, 2024-02-04:
HI, i encountered the same situation, compiles well, but can't run or debug due to:
```
[02:42:16:568 - 7F96F7C09140]: warning: allocate 28 bytes memory failed
Exception: allocate memory failed
```
The difference is that my platform is in a Linux environment, tried to modify it and increase Max linear memory size. but it doesn't seem to work, and wasm-toolchain, wasm-debug-server and VSCode extension all are version 1.3.0.

TianlongLiang, 2024-02-22:
Hi, sorry for the late reply, I think after [this pr](https://github.com/bytecodealliance/wasm-micro-runtime/pull/3172). You can rebuild VS Code extension `wamride-1.3.2.vsix` and docker image `wasm-debug-server`:
```shell
# if you already have this version of docker image
docker rmi wasm-debug-server:1.3.2
# in directory test-tools/wamr-ide/WASM-Debug-Server/Docker
docker build -t wasm-debug-server:1.3.2 .
# in directory test-tools/wamr-ide/VSCode-Extension
vsce package
```

And you should see a `wamride-1.3.2.vsix` under `test-tools/wamr-ide/VSCode-Extension` directory. You can delete the existing extension and manually load this one. After that, it should automatically download lldb and other component needed. Then it should work now and won't have that message.

HoyiTT, 2024-04-16:
I had a similar problem. 

``warning: allocate 16 bytes memory failed
Exception: allocate memory failed``

But now it's working.
Add a heap size in the iwasm run option.

Like `./iwasm --heap-size=16384  test.wasm`

I hope it helps you