wasm-micro-runtime/wasm-micro-runtime #2593: Different results while executing the same wasm file with different runtime tools

luxinyi0105, 2023-09-25:
### Describe the bug
While executing the given testcase with wasm-micro-runtime and other runtime tools, the results are different.

### Test Case
[The given testcase](https://github.com/luxinyi0105/wasm_runtime_test/raw/main/09-25.zip) was simply mutated a wasm file, which was obtained by compiling C program generated with `Csmith` using `Emscripten Compiler`(Emcc). 

The original C program is `c_file.c`, the compilation results with Emscripten is `wasm_file.wasm`, and its wat format is `wat_file.wat`. 

We mutated the wat file to change its `i32.add` in line 11788 with `i32.sub`, and change all the `i32/i64.shl` with `i32/i64.xor`. The result after mutation is `mutated_file.wat`, and its wasm format is `mutated_file.wasm`.

### Expected Results
```sh
$ iwasm mutated_file.wasm
checksum = CF2196F3
```

### Actual Results
```sh
$ iwasm mutated_file.wasm
checksum = CF2196F3
Exception: out of bounds memory access
```

### Versions and Environment
Tools version: iwasm 1.2.3
Operating system: Ubuntu 22.04.1
Architecture: x86_64

### Extra Info
I tried to use some other wasm runtimes, such as wasmer, wasmtime, wasmedge and wasmi to execute the same testcase, wasmtime outputs checksum result with runtime error caused by misaligned pointer, while other three tools just output checksum result without runtime error. 

I'm not sure whether the runtime error caused by OOB here is correct. Or maybe wasm-micro-runtime should also output runtime error caused by misaligned pointer like wasmtime? The detail still needs you to confirm. Thanks a lot!

wenyongh, 2023-09-26:
Hi, I checked the running details in both WAMR and wasmedge, after outputing `checksum = CF2196F3`, they both ran into the native wrapper of the libc-wasi fd_write again with invalid address to access the linear memory:

https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/libraries/libc-wasi/libc_wasi_wrapper.c#L660-L663

https://github.com/WasmEdge/WasmEdge/blob/master/lib/host/wasi/wasifunc.cpp#L1074-L1078
(IOV.buf = 4864, IOV.buf_len = 17165056, while the linear memory size is 16777216)

The difference is that WAMR throws the "out of bounds memory access" exception while wasmedge doesn't, instead it returns `__WASI_ERRNO_FAULT` error. I think how to handle the invalid address here depends on runtime, e.g. wasm3 throws exception also.

luxinyi0105, 2023-09-26:
> (IOV.buf = 4864, IOV.buf_len = 17165056, while the linear memory size is 16777216)

Thanks for your reply, and I wonder how wamr deal with `misaligned pointer` when execute a wasm file? 

In [wasmtime issue #7081](https://github.com/bytecodealliance/wasmtime/issues/7081), I have communicated with wasmtime developer, and confirmed that there is indeed a problem of misaligned pointer in the given testcase. 
In issue #2396, I have committed similar issue but have not received a clear response yet.

wenyongh, 2023-09-26:
IIUC, the check for misaligned pointer is the new feature in witx2, and currently it is supported by wasmtime. Most of the runtimes (including WAMR) doesn't support it yet.