wasm-micro-runtime/wasm-micro-runtime #2616: Illegal Instruction on valid WASM code with AddressSanitizer

khagankhan, 2023-10-04:
## Describe the Bug

When executing a valid WebAssembly module using the `iwasm`, the program terminates with a "Illegal Instruction" or "AddressSanitizer:DEADLYSIGNAL"

`iwasm --version`: iwasm 1.2.3

`uname -a`: Linux node0.khan22-169985.randtest.emulab.net 5.15.0-56-generic #62-Ubuntu SMP Tue Nov 22 19:54:14 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

## Steps to reproduce:
1. Compile the code with AddressSanitizer enabled:
```
AFL_USE_ASAN=1 CC=/usr/local/bin/afl-clang-fast CXX=/usr/local/bin/afl-clang-fast++ cmake .. \
    -DCMAKE_C_COMPILER_RANLIB=/usr/bin/ranlib \
    -DCMAKE_CXX_COMPILER_RANLIB=/usr/bin/ranlib \
    -DCMAKE_C_COMPILER_AR=/usr/bin/ar \
    -DCMAKE_CXX_COMPILER_AR=/usr/bin/ar \
    -DCMAKE_C_COMPILER_LAUNCHER=/usr/local/bin/afl-clang-fast \
    -DCMAKE_C_COMPILER_AR=/usr/bin/ar  \
    -DCMAKE_CXX_COMPILER_LAUNCHER=/usr/local/bin/afl-clang-fast++ \
    -DCMAKE_C_FLAGS="-fsanitize=address,leak,undefined -g" \
    -DCMAKE_CXX_FLAGS="-fsanitize=address,leak,undefined -g"
```

1. Save the `illegal.wat` (Remove .txt extension)
2. Install and build WABT toolkit (`wat2wasm` from WABT can be used to convert `illegal.wat` into `illegal.wasm` using: `wat2wasm illegal.wat -o illegal.wasm`. I have also added `illegal.wasm` that obviates the need to use `wat2wasm`)
3. Run `iwasm illegal.wasm`
4. Observe the `Illegal instruction` or `AddressSanitizer bug`.

## Files that cause the issue:
[illegal.wat.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/12801240/illegal.wat.txt) (Remove .txt extension when save the file)

[illegal.wasm.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/12801241/illegal.wasm.txt) (Remove .txt extension when save the file)

## Observed Behavior:
```
iwasm illegal.wasm 

=================================================================
AddressSanitizer:DEADLYSIGNAL
AddressSanitizer: nested bug in the same thread, aborting.
```

## Expected behavior:
`iwasm` is supposed to run the module successfully. For example, `iwasm`, compiled the same way runs this module successfully.

## GDB Backtrace

```
(gdb) run illegal.wasm 
Starting program: /root/wasmoi/targets/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm illegal.wasm
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0x00005555557514da in wasm_loader_prepare_bytecode (module=<optimized out>, func=0x60b0000001a0, cur_func_idx=<optimized out>, error_buf=<optimized out>, error_buf_size=<optimized out>) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:7300
7300	                    bh_memcpy_s(loader_ctx->frame_offset, size,
(gdb) bt
#0  0x00005555557514da in wasm_loader_prepare_bytecode (module=<optimized out>, func=0x60b0000001a0, 
    cur_func_idx=<optimized out>, error_buf=<optimized out>, error_buf_size=<optimized out>)
    at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:7300
#1  0x0000555555734ddd in load_from_sections (module=<optimized out>, sections=0x603000000130, 
    is_load_from_file_buf=<optimized out>, error_buf=<optimized out>, error_buf_size=<optimized out>)
    at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:3553
#2  0x000055555573c6f0 in load (buf=<optimized out>, size=<optimized out>, module=0x613000000040, 
    error_buf=0x555555d43680 <__afl_area_initial> "", error_buf_size=128)
    at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:3913
#3  wasm_loader_load (buf=<optimized out>, size=<optimized out>, error_buf=<optimized out>, error_buf_size=<optimized out>)
    at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:4080
#4  0x000055555564ae59 in main (argc=1, argv=<optimized out>)
    at /root/wasmoi/targets/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:916
```

### The code that caused the issue (`cat illegal.wat`):

```
(module (memory $mem 1) (table 1 funcref)
   (elem (i32.const 0) $lift_1)
   (func
    $main
    (export "_main")
    (result i32)
    (local $i32_storage i32)
    (local $i64_storage i64)
    (local $f32_storage f32)
    (local $f64_storage f64)
    i32.const
    737
    i64.load32_s
    offset=28
    align=1
    i32.const
    0
    call_indirect
    (param i64)
    (result f32)
    i32.reinterpret_f32)
   (func
    $lift_1
    (param $param_2 i64)
    (result f32)
    (local $i32_storage i32)
    (local $i64_storage i64)
    (local $f32_storage f32)
    (local $f64_storage f64)
    i32.const
    129
    i64.load16_u
    offset=1
    align=1
    i32.const
    26
    i64.load16_s
    offset=62
    align=1
    i64.ge_s
    i32.const
    2
    i32.shr_u
    f32.reinterpret_i32
    f32.neg
    f32.trunc))
```
### An example code that works/unable to found the issue: (`cat working.wat`:)
[working.wat.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/12801271/working.wat.txt) (Remove .txt when to save the file)
[working.wasm.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/12801272/working.wasm.txt) (Remove .txt when to save the file)

```
(module (memory $mem 1) (table 0 funcref)
   (elem (i32.const 0))
   (func
    $main
    (export "_main")
    (result i32)
    (local $i32_storage i32)
    (local $i64_storage i64)
    (local $f32_storage f32)
    (local $f64_storage f64)
    i32.const
    796
    i32.load8_u
    offset=75
    align=1))
```

## Additional information

A combination of AFL++ and Wasmlike, an Xsmith-based random program generator produced the snippet of code that found the issue. [Xsmith Project](https://www.flux.utah.edu/project/xsmith)

wenyongh, 2023-10-23:
@khagankhan many thanks for using the sanitizer tool to test iwasm, I tried building iwasm with your command and running the case `illegal.wasm` many times, for the below error reported:
```bash
AddressSanitizer:DEADLYSIGNAL
AddressSanitizer: nested bug in the same thread, aborting.
```
I found it was caused by function `touch_pages` in posix_thread.c:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/common/posix/posix_thread.c#L449-L465

The function calls `os_alloca` many times to allocate memory from native stack and is not allowed to be tested with sanitizer, you can see that `__attribute__((no_sanitize_address))` was added for the gcc compiler, which it is not added for the clang compiler. After I added `__attribute__((no_sanitize_address))` for the clang comiler, I didn't find the error again:
```diff
--- a/core/shared/platform/common/posix/posix_thread.c
+++ b/core/shared/platform/common/posix/posix_thread.c
@@ -443,6 +443,7 @@ static os_thread_local_attribute uint8 *sigalt_stack_base_addr;

 #if defined(__clang__)
 #pragma clang optimize off
+__attribute__((no_sanitize_address))
 #elif defined(__GNUC__)
```
I am not sure how to reproduce the below error with gdb:
```
Program received signal SIGILL, Illegal instruction.
0x00005555557514da in wasm_loader_prepare_bytecode (module=<optimized out>, func=0x60b0000001a0, cur_func_idx=<optimized out>, error_buf=<optimized out>, error_buf_size=<optimized out>) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:7300
7300	                    bh_memcpy_s(loader_ctx->frame_offset, size,
```
Could you try the patch above and test again? Thanks.

khagankhan, 2023-10-23:
Hi! It seems to stem from the previous lack of `+__attribute__((no_sanitize_address))`. With `gcc` and `clang` it seems to be patched! Thank you for detailed explanation!

Best. 

 Khan

khagankhan, 2023-10-26:
Hello @wenyongh! Thank you discussing this further! I have tried to rerun it with `gcc` and `clang`. It seems to be fixed!

Thank you!

wenyongh, 2024-01-11:
@khagankhan welcome.

I close this issue since it was fixed😊

closed 2024-01-11 by wenyongh