wasm-micro-runtime/wasm-micro-runtime #2617: Potential Control Flow Integrity Violation in wasm_runtime_malloc_internal

khagankhan, 2023-10-05:
## Describe the Bug

The `wasm_runtime_malloc_internal` function may be vulnerable to control flow integrity violations, potentially due to the misuse or mismanagement of the `malloc_func` function pointer.

`iwasm --version`: iwasm 1.2.3

`uname -a`: Linux node2.khan22-169985.randtest.emulab.net 5.15.0-56-generic #62-Ubuntu SMP Tue Nov 22 19:54:14 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

## Steps to reproduce:
1. Compile the code with CFISAN enabled:
```
    -DCMAKE_C_COMPILER_RANLIB=/usr/bin/ranlib \
    -DCMAKE_CXX_COMPILER_RANLIB=/usr/bin/ranlib \
    -DCMAKE_C_COMPILER_AR=/usr/bin/ar \
    -DCMAKE_CXX_COMPILER_AR=/usr/bin/ar \
    -DCMAKE_C_COMPILER_LAUNCHER=/usr/local/bin/afl-clang-fast \
    -DCMAKE_C_COMPILER_AR=/usr/bin/ar  \
    -DCMAKE_CXX_COMPILER_LAUNCHER=/usr/local/bin/afl-clang-fast++ \
    -DCMAKE_C_FLAGS="-fsanitize=cfisan-g" \
    -DCMAKE_CXX_FLAGS="-fsanitize=cfisan -g"
```
2. Save the `illegal.wat` (Remove .txt extension)
3. Install and build WABT toolkit (`wat2wasm` from WABT can be used to convert `illegal.wat` into `illegal.wasm` using: `wat2wasm illegal.wat -o illegal.wasm`. I have also added `illegal.wasm` that obviates the need to use `wat2wasm`)
4. Run `iwasm illegal.wasm`
5. Observe the `Illegal instruction`.

## Files that cause the issue:
[illegal.wat.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/12811314/illegal.wat.txt)

[illegal.wasm.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/12811315/illegal.wasm.txt)

## Observed Behavior:
Potential control flow integrity violation if `malloc_func` is not properly managed or if it's tampered with.
```
iwasm illegal.wasm 
Illegal instruction
```

## Expected behavior:
The `malloc_func` should always point to a valid memory allocation function and should not be overwritten or tampered with by external or unexpected means.

## GDB Backtrace

```
Reading symbols from ../targets/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm...
(gdb) break main
Breakpoint 1 at 0x9ae0: file /root/wasmoi/targets/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c, line 540.
(gdb) run illegal.wasm 
Starting program: /root/wasmoi/targets/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm illegal.wasm
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Breakpoint 1, main (argc=2, argv=0x7fffffffe418) at /root/wasmoi/targets/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:540
540	{
(gdb) c
Continuing.

Program received signal SIGILL, Illegal instruction.
0x0000555555567cb9 in wasm_runtime_malloc_internal (size=32) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_memory.c:177
177	        return malloc_func(size);
(gdb) bt
#0  0x0000555555567cb9 in wasm_runtime_malloc_internal (size=32) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_memory.c:177
#1  wasm_runtime_malloc (size=32) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_memory.c:253
#2  0x000055555555e3dd in register_natives (n_native_symbols=52, call_conv_raw=false, module_name=<optimized out>, native_symbols=<optimized out>)
    at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_native.c:337
#3  wasm_native_register_natives (n_native_symbols=52, module_name=<optimized out>, native_symbols=<optimized out>)
    at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_native.c:377
#4  wasm_native_init () at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_native.c:573
#5  wasm_runtime_env_init () at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:432
#6  wasm_runtime_full_init (init_args=<optimized out>) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:672
#7  main (argc=<optimized out>, argv=<optimized out>) at /root/wasmoi/targets/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:871
```

### The code whose execution firstly found the issue (`cat illegal.wat`):

```
(module (memory $mem 1) (global $lift_1 (mut i32) (i32.const 346))
   (table 0 funcref)
   (elem (i32.const 0))
   (func
    $main
    (export "_main")
    (result i32)
    (local $i32_storage i32)
    (local $i64_storage i64)
    (local $f32_storage f32)
    (local $f64_storage f64)
    i32.const
    468
    i32.load16_s
    offset=13
    align=1
    global.get
    $lift_1
    i32.and
    global.set
    $lift_1
    global.get
    $lift_1
    i32.popcnt))
```

## Additional information

A combination of AFL++ and Wasmlike, an Xsmith-based random program generator produced the snippet of code that found the issue. [Xsmith Project](https://www.flux.utah.edu/project/xsmith)

wenyongh, 2023-10-23:
@khagankhan Could you upload the whole command to build iwasm? I used the below command:
```bash
AFL_USE_ASAN=1 CC=/usr/bin/afl-clang-fast CXX=/usr/bin/afl-clang-fast++ cmake .. \
-DCMAKE_C_COMPILER_RANLIB=/usr/bin/ranlib \
-DCMAKE_CXX_COMPILER_RANLIB=/usr/bin/ranlib \
-DCMAKE_C_COMPILER_AR=/usr/bin/ar \
-DCMAKE_CXX_COMPILER_AR=/usr/bin/ar \
-DCMAKE_C_COMPILER_LAUNCHER=/usr/bin/afl-clang-fast \
-DCMAKE_CXX_COMPILER_LAUNCHER=/usr/bin/afl-clang-fast++ \
-DCMAKE_C_FLAGS="-fsanitize=cfisan-g" \
-DCMAKE_CXX_FLAGS="-fsanitize=cfisan-g"
```
But got error:
```bash
  CMake Error at /usr/share/cmake-3.22/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler
    "/usr/bin/afl-clang-fast"
  is not able to compile a simple test program.
  ....
  clang: error: unsupported argument 'cfisan-g' to option 'fsanitize='
```

khagankhan, 2023-10-26:
Hey @wenyongh! Sorry for that. It is because I had typo in the provided instrumentation. 

In the following code `cfisan` should be changed to `cfi`

```
-DCMAKE_C_FLAGS="-fsanitize=cfisan-g" \
-DCMAKE_CXX_FLAGS="-fsanitize=cfisan-g"
```

So it should be:
1. `export AFL_USE_CFISAN=1`

2.
```
 CC=/usr/local/bin/afl-clang-fast CXX=/usr/local/bin/afl-clang-fast++ cmake .. \
    -DCMAKE_C_COMPILER_RANLIB=/usr/bin/ranlib \
    -DCMAKE_CXX_COMPILER_RANLIB=/usr/bin/ranlib \
    -DCMAKE_C_COMPILER_AR=/usr/bin/ar \
    -DCMAKE_CXX_COMPILER_AR=/usr/bin/ar \
    -DCMAKE_C_COMPILER_LAUNCHER=/usr/local/bin/afl-clang-fast \
    -DCMAKE_C_COMPILER_AR=/usr/bin/ar  \
    -DCMAKE_CXX_COMPILER_LAUNCHER=/usr/local/bin/afl-clang-fast++ \
    -DCMAKE_C_FLAGS="-fsanitize=cfi -g" \
    -DCMAKE_CXX_FLAGS="-fsanitize=cfi -g"
```

After getting `Illegal Instruction`, you can observe it on `gdb`:

```
root@node0:~/wasmoi/targets/wasm-micro-runtime/product-mini/platforms/linux/build# gdb ./iwasm 
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./iwasm...
(gdb) break main
Breakpoint 1 at 0x9ae0: file /root/wasmoi/targets/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c, line 540.
(gdb) run illegal.wasm 
Starting program: /root/wasmoi/targets/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm illegal.wasm
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Breakpoint 1, main (argc=2, argv=0x7fffffffe358) at /root/wasmoi/targets/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:540
540	{
(gdb) c
Continuing.

Program received signal SIGILL, Illegal instruction.
0x0000555555567ca9 in wasm_runtime_malloc_internal (size=32) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_memory.c:177
177	        return malloc_func(size);
(gdb) bt
#0  0x0000555555567ca9 in wasm_runtime_malloc_internal (size=32) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_memory.c:177
#1  wasm_runtime_malloc (size=32) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_memory.c:253
#2  0x000055555555e3dd in register_natives (n_native_symbols=52, call_conv_raw=false, module_name=<optimized out>, native_symbols=<optimized out>) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_native.c:337
#3  wasm_native_register_natives (n_native_symbols=52, module_name=<optimized out>, native_symbols=<optimized out>) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_native.c:377
#4  wasm_native_init () at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_native.c:572
#5  wasm_runtime_env_init () at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:430
#6  wasm_runtime_full_init (init_args=<optimized out>) at /root/wasmoi/targets/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:670
#7  main (argc=<optimized out>, argv=<optimized out>) at /root/wasmoi/targets/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:871
(gdb) info registers
rax            0x1d8               472
rbx            0x0                 0
rcx            0x5555555f14b0      93824992875696
rdx            0x7fffffffe301      140737488347905
rsi            0x20                32
rdi            0x20                32
rbp            0x7fffffffe636      0x7fffffffe636
rsp            0x7fffffffdee0      0x7fffffffdee0
r8             0x7fffffffe160      140737488347488
r9             0x0                 0
r10            0x7ffff7fc3908      140737353890056
r11            0x7ffff7fde660      140737353999968
r12            0x0                 0
r13            0x7fffffffe360      140737488348000
r14            0x0                 0
r15            0x2                 2
rip            0x555555567ca9      0x555555567ca9 <wasm_runtime_malloc+329>
eflags         0x10202             [ IF RF ]
cs             0x33                51
ss             0x2b                43
ds             0x0                 0
es             0x0                 0
fs             0x0                 0
gs             0x0                 0

```

I am ready to provide any clarification and cooperate to fix the issue if have any!

Best

 Khan

wenyongh, 2023-11-02:
Hi, the issue is due to that the prototypes of system malloc function and the malloc function for runtime allocator are different, the former's argument is `size_t`, while the latter is `unsigned int`, I am not sure why in the sanitizer, passing `unsigned int` argument to the former is not allowed. It ran good after applying the patch below:
```diff
diff --git a/product-mini/platforms/posix/main.c b/product-mini/platforms/posix/main.c
index 09bf7c55..5c72aeb6 100644
--- a/product-mini/platforms/posix/main.c
+++ b/product-mini/platforms/posix/main.c
@@ -538,6 +538,36 @@ timeout_thread(void *vp)
 }
 #endif

+static void *
+malloc_func(
+#if WASM_MEM_ALLOC_WITH_USER_DATA != 0
+    void *user_data,
+#endif
+    unsigned int size)
+{
+    return malloc(size);
+}
+
+static void *
+realloc_func(
+#if WASM_MEM_ALLOC_WITH_USER_DATA != 0
+    void *user_data,
+#endif
+    void *ptr, unsigned int size)
+{
+    return realloc(ptr, size);
+}
+
+static void
+free_func(
+#if WASM_MEM_ALLOC_WITH_USER_DATA != 0
+    void *user_data,
+#endif
+    void *ptr)
+{
+    free(ptr);
+}
+
 int
 main(int argc, char *argv[])
 {
@@ -861,9 +891,9 @@ main(int argc, char *argv[])
     init_args.mem_alloc_option.pool.heap_size = sizeof(global_heap_buf);
 #else
     init_args.mem_alloc_type = Alloc_With_Allocator;
-    init_args.mem_alloc_option.allocator.malloc_func = malloc;
-    init_args.mem_alloc_option.allocator.realloc_func = realloc;
-    init_args.mem_alloc_option.allocator.free_func = free;
+    init_args.mem_alloc_option.allocator.malloc_func = malloc_func;
+    init_args.mem_alloc_option.allocator.realloc_func = realloc_func;
+    init_args.mem_alloc_option.allocator.free_func = free_func;
 #endif

 #if WASM_ENABLE_FAST_JIT != 0
```

wenyongh, 2023-11-03:
Refer to https://github.com/bytecodealliance/wasm-micro-runtime/pull/2717

closed 2023-11-03 by wenyongh

khagankhan, 2023-11-03:
Works good! Thanks for your prompt response!

PR #2717 (2023-11-03): Use user defined malloc/free functions for user defined memory allocator
Fixes #2617.