wasm-micro-runtime/wasm-micro-runtime #2631: Fast jit dump output is out of order

Rlyown, 2023-10-11:
Hi,

I normally try to observe the Fast Jit process through `-DWAMR_BUILD_FAST_JIT_DUMP=1`, and set macro `#define CODEGEN_DUMP 1`. But I found that its printed content is out of order . 

Maybe it has a problem with multi-threaded output because of the `orcjit_thread_callback` function. Is there any good way to view the dump output?

A part of output content looks like this. 
```text
, L3I16I40, 0x158:
    ; PREDS(
    LDI32          )
    ; BEGIN_BCIP=0x0000,  END_BCIP=0x0000 0x4
    ADD            i13, I21
,  I35,     LDPTR          0x10i15 I8, , I15i11, 0x10
,     STI32
 i15, I16, 0x1a8
    LDPTR           I42
    U32TOI64        I36, i11
    ADD             I37, 0x10L0x10, I36
    STI32           i13, I21,
,     CALLNATIVE     I0, 0x0 VOID, I370x555b27a6ac10L
    U32TOI64       ,
I8    CALLNATIVE         LDPTR           I43, I42 , VOID0x30I38, ,
0x555b27a6ac10Li12,     ADD            I40 I43, I43, 0x10
, , 0x10    RETURN
    RETURN
 0x1    ADD             0x10x4L
I39    STPTR          ,  I43, I42, 0x14L
, I38    ; SUCCS(
0x30)
```

By the way, this is my complete cmake command, in order to enable `FAST JIT` only.
```sh
cmake .. -DWAMR_BUILD_INTERP=0 -DWAMR_BUILD_AOT=0 -DWAMR_BUILD_JIT=0 -DWAMR_BUILD_FAST_JIT=1 -DCMAKE_EXPORT_COMPILE_COMMANDS=1 -DCODEGEN_DUMP=1 -DWAMR_BUILD_FAST_JIT_DUMP=1 
```

TianlongLiang, 2023-10-11:
Hi, I encountered this problem before, and my solution(if I remember correctly) is to modify [this line](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/config.h#L90C1-L90C1), set 
```C
#define WASM_ORC_JIT_BACKEND_THREAD_NUM 1
```
Let me know if that helps, if not, I will dig deeper into my memory to find the solution

Rlyown, 2023-10-11:
@TianlongLiang It works. Thank you!

closed 2023-10-11 by Rlyown