wasm-micro-runtime/wasm-micro-runtime #2635: wasi-threads tests not working with fast interpreter

christinaings, 2023-10-11:
Hello,

I am one of the Amazon devs working on wasi-threads and we are trying to use [wasm parallel gzip](https://github.com/bytecodealliance/wasm-parallel-gzip/tree/main) to test performance of wasi-threads on devices like NVIDIA Shield. I'm encountering an error when running this with the fast interpreter, namely:

```
2023-10-09 17:39:46.607 11748-11832 wasm_runtime::          com...groom.ignite.testrunner.debug  I  [04:09:37:726 - 943811C0]: 
2023-10-09 17:39:46.608 11748-11832 wasm_runtime::          com...groom.ignite.testrunner.debug  I  WASM module load failed: pre-compiled label offset out of range
```

I have run this with the classic interpreter and do not encounter the issue. Does anyone have any suggestions on how this can be resolved? I am unable to attach the wasm file, as wasm file type is not supported.

wenyongh, 2023-10-20:
Hi, this may be caused by too much code size in fast interpreter, not sure whether you compiled iwasm with release mode, and did you enable some unnecessary features? If release mode and only necessary features are enabled, could you try disabling the labels-as-values feature: in core/config.h, change `#define WASM_ENABLE_LABELS_AS_VALUES 1` to `#define WASM_ENABLE_LABELS_AS_VALUES 0`, or add `add_definitions(-DWASM_ENABLE_LABELS_AS_VALUES=0)` in CMakeLists.txt.

cimacmillan, 2023-10-20:
Hi @wenyongh, I think the cause is due to being built in debug mode like you mention. Where it's failing the check before emitting the function address for bytecode dispatch [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/interpreter/wasm_loader.c#L5396). The opcode is 0x10 and offset becomes 55972 in debug mode. In release build it's 23748. It is ARMv7 with eabi.

For debug mode, I've set `WASM_ENABLE_LABELS_AS_VALUES 0` as I think that will be fine for our use-case, and the test we have is passing now. 

I wonder what is a resolution to this, and whether it's a long-term risk as more opcodes are added, or opcode handler implementations extended, even for release build. I notice with the implementation it's interpreting the function address as int8 pointer. I thought depending on native instruction size, you could interpret this as a int16 or int32 ptr, then when doing the dispatch to map this back so it dispatches to the right address.  Eg on this processor, all the function addresses are multiples of 4 when interpreted as int8 pointer, so I guess interpreting it as int32 pointer would mean the Debug compiled dispatch table would then fit in the instruction. I'm not sure about this though, so would take your opinion.

wenyongh, 2023-10-23:
@cimacmillan thanks for the reminding, yes, the issue may often occur if more opcodes are added, using 32-bit offset/ptr is better, I submitted PR #2659 to fix it, please try again.

wenyongh, 2024-01-05:
Close this issue as it was fixed.

closed 2024-01-05 by wenyongh