wasm-micro-runtime/wasm-micro-runtime #2641: Different behavior when running llvm-jit and fast-jit

unicornt, 2023-10-15:
Hi, I build `wamr 1.2.3` on ubuntu 20.04 and found a different behavior when running the same code with `llvm-jit` mode and `fast-jit` mode.
`fast-jit` mode will exit normally while `llvm-jit` mode throw a exception `Exception: integer divide by zero`

I also use other runtimes to run the same code, and it looks like a wrong behavior of `llvm-jit` mode

By the way, I try to replace all `div` in wasm code with `add`. But it turns out to be **useless**.

[code.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12908730/code.zip)

[image]

TianlongLiang, 2023-10-16:
Hi, I tested it on the latest main branch, it seems that this issue no longer exists on the latest commit

unicornt, 2023-10-16:
Thank you! Maybe you can tell me why it happened?

TianlongLiang, 2023-10-16:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/2627 This PR fixes the problem, it has something to do with the LLVM const( in some scenarios the constant might not be directly available to the LLVM IR builder)

closed 2023-10-16 by unicornt