wasm-micro-runtime/wasm-micro-runtime #2642: Abnormal long execute time running wasm code with wamr

unicornt, 2023-10-16:
Hi, I found a case that may cause an abnormal long executing time problem on wamr.

### Test Case
[code.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12912501/report1.zip)

### Results
 I found that `wamr`(both `llvm-jit` mode and `fast-jit` mode) takes a much longer time than other runtimes, including `wasmedge`(AOT mode) and `wasmtime`.

[image]

### Versions and Environment

wamr version : 1.2.3

Operating system: Ubuntu 20.04

Architecture: x86_64

wenyongh, 2023-10-20:
Hi, this issue is similar to issue #2602, and if we apply the patch #2606 (or as below), iwasm works like the runtimes you mentioned. But after discussion, we decided not fix it, please refer to #2602 and #2606 for more details.

```diff
--- a/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c
+++ b/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c
@@ -730,6 +730,13 @@ fd_table_insert_existing(struct fd_table *ft, __wasi_fd_t in, int out)
 #endif
     }

+    if (in == 0)
+        /* Strip off write bit for stdin */
+        rights_base &= ~(__wasi_rights_t)__WASI_RIGHT_FD_WRITE;
+    else if (in == 1 || in == 2)
+        /* Strip off read bit for stdout and stderr */
+        rights_base &= ~(__wasi_rights_t)__WASI_RIGHT_FD_READ;
+
```

unicornt, 2023-10-20:
Interesting discussion! I also think we should `not` keep file descriptor of `stdin` be read-only.
Thank you for the reply!

closed 2023-10-20 by unicornt