wasm-micro-runtime/wasm-micro-runtime #2643: how to build wasm from rust with shared-memory support

zhen9910, 2023-10-16:
I tried to a wasm module from C and Rust with shared-memory support, and was able to build it from C with  below cmd and options.

```
/opt/wasi-sdk/bin/clang     \
    --target=wasm32-wasi -v -O0 -z stack-size=32768 -Wl,--initial-memory=65536 \
    --sysroot=/opt/wasi-sdk/share/wasi-sysroot  \
    -Wl,--strip-all,--no-entry -nostdlib -pthread \
    -Wl,--shared-memory,--max-memory=131072      \
    -Wl,--export=foo\
    -Wl,--export=__heap_base,--export=__data_end    \
    -Wl,--export=__wasm_call_ctors  \
    -Wl,--allow-undefined \
    -o ${WASM_FILE_NAME} main.c
```

But I got some problem to build it from rust. 
I converted the above cmd options to rustflags as blow, but it looks cargo doesn't know codegen options "sysroot", "-pthread", and "-nostdlib".
Can you suggest how to set these flags for rust?

```
[build]
rustflags = [
  "-C", "link-arg=-zstack-size=32768",
  "-C", "link-arg=--initial-memory=65536",
  "-C", "link-arg=--strip-all",
  "-C", "link-arg=--no-entry",
  "-C", "sysroot=/opt/wasi-sdk/share/wasi-sysroot",
  "-C", "-pthread",
  "-C", "-nostdlib",
  "-C", "link-arg=--shared-memory",
  "-C", "link-arg=--max-memory=131072",
  "-C", "link-arg=--export=foo",
  "-C", "link-arg=--export=__heap_base",
  "-C", "link-arg=--export=__data_end",
  "-C", "link-arg=--export=__wasm_call_ctors",
  "-C", "link-arg=--allow-undefined",
]
```

wenyongh, 2023-10-20:
Hi, maybe `link-arg=--shared-memory` and some others are enough to enabled the shared memory, could you try removing the following flags:
```
  "-C", "link-arg=--no-entry",
  "-C", "sysroot=/opt/wasi-sdk/share/wasi-sysroot",
  "-C", "-pthread",
  "-C", "-nostdlib",
```

zhen9910, 2023-10-20:
Hi, thanks for your suggestion, but this actually gives another error, 

```
  = note: rust-lld: error: --shared-memory is disallowed by std-056ca9b4df39b1a5.std.078b28f0-cgu.0.rcgu.o because it was not compiled with 'atomics' or 'bulk-memory' features.
          rust-lld: error: 'atomics' feature must be used in order to use shared memory
       
```

it looks ```link-arg=--shared-memory``` has to be used together with multi-thread enabled, but not sure how to add -pthread flag to rustflag.

```"-C", "-pthread",``` or ```"-C", "link-arg=-pthread",``` is not allowed with error ```error: unknown codegen option: `-pthread` ``` or ``` = note: rust-lld: error: unknown argument: -pthread ```

wenyongh, 2023-10-21:
Could you try adding: `"-C", "target-feature=+atomics,+mutable-globals,+bulk-memory"?`
Refer to https://github.com/Amanieu/parking_lot/issues/308