wasm-micro-runtime/wasm-micro-runtime #2645: Different behavior when running the same wasm file

unicornt, 2023-10-16:
Hi, I found a case that will cause different behavior when running the same wasm file. 
However, I'm not sure whether it is a bug or not. I think maybe it is caused by the memory strategy wamr has. So I asked you developers for help. Thanks a lot!

### Test Case

[code.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12913702/report2.zip)

### Result

This code can exit normally on other Wasm Runtimes (wasmedge, wasmtime)
But it can **not** exit even after running 60 seconds on wamr (both fast-jit mode and llvm-jit mode)

[image]

I also found that if I change the path of this file, or change its file name. Wamr may throw an exception like `out of bounds memory access` or `integer overflow`.

[image]
[image]

### Extra information
This code calls wasi function `args_get`.

### Versions and Environment

wamr version : 1.2.3

Operating system: Ubuntu 20.04

Architecture: x86_64

lum1n0us, 2023-10-18:
Initial analysis shows *iwasm* is involved in a loop which includes `fd_write()`. It would be very helpful with the source code of *case.wasm*. would u mind?

unicornt, 2023-10-18:
> Initial analysis shows _iwasm_ is involved in a loop which includes `fd_write()`. It would be very helpful with the source code of _case.wasm_. would u mind?

Sorry, `case.wasm` is generated directly which means there is no source code.
By the way, in which loop? Maybe I can help to explain how it is generated.

lum1n0us, 2023-10-19:
It seems iwasm stucks in `func[11]`

```
`func[5] <_start>`
  -> `func[4]`
  -> `func[0]` <wasi_snapshot_preview1.args_sizes_get>
  -> `func[1]` <wasi_snapshot_preview1.args_get>
  -> `func[30]`
    -> `func[32]`
    -> `func[32]`
    -> `func[31]`
  -> `func[30]`
    -> `func[31]`
  -> `func[9]`
    -> `func[20]`
      -> `func[14]`
      -> `func[21]`
      -> `func[17]`
  -> `func[9]`
    -> `func[20]`
      -> `func[14]`
      -> `func[21]`
      -> `func[17]`
      -> `func[11]`
        -> `func[3]` <wasi_snapshot_preview1.fd_write>
        -> `func[28]`
        -> `func[3]` <wasi_snapshot_preview1.fd_write>
        -> `func[28]`
        -> ...
```

There is loop looks like:

```
0013ce func[11]:
 ...
 0014a1: 03 40                      |         loop
 0014a3: 20 06                      |           local.get 6
 0014a5: 20 03                      |           local.get 3
 ...
 0014ba: 0d 02                      |           br_if 2       <- stack top: 0x0
 ...
 0014c1: 04 40                      |           if            <- stack top: 0x0
 0014c3: 20 04                      |             local.get 4
 0014c5: 21 05                      |             local.set 5
 0014c7: 0c 04                      |             br 4
 0014c9: 0b                         |           end
 ...
 001546: 10 03                      |           call 3 <wasi_snapshot_preview1.fd_write>
 001548: 10 1c                      |           call 28                                 
 00154a: 45                         |           i32.eqz     <- stack top: 0x0
 00154b: 41 01                      |           i32.const 1
 00154d: 71                         |           i32.and     <- stack top: 0x1, 0x1
 00154e: 20 06                      |           local.get 6
 001550: 41 01                      |           i32.const 1
 001552: 71                         |           i32.and    <- stack top: 0x17fffff7, 0x00000001
 001553: 72                         |           i32.or     <- stack top: 0x00000001, 0x00000001
 001554: 0d 00                      |           br_if 0     <- stack top: 0x1
 001556: 0b                         |         end
```

Plus `call 3` is trying to write 2 bytes into `fd 0` (stdin).

unicornt, 2023-10-19:
I believe `func 11` is generated based on `__stdio_write` function. We add some wasm code directly into it. Maybe that's why it leads to an abnormal access to memory or wasi.
Here is the original .wat code of `__stdio_write`.
[func.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13038023/func.wat.zip)

lum1n0us, 2023-10-19:
*__MACOSX/._func.wat* seems not a regular WAT file. it is a "__MACOSX/._func.wat: AppleDouble encoded Macintosh file".

unicornt, 2023-10-19:
> ___MACOSX/._func.wat_ seems not a regular WAT file. it is a "__MACOSX/._func.wat: AppleDouble encoded Macintosh file".

I only copy `__stdio_write` function to `func.wat`. You can compare `func.wat` and` func 11` to see what code is added.

lum1n0us, 2023-10-19:
I am wondering would u mind describing the expected output of case.wasm since other runtime, like wasmtime, will just return quitely.

unicornt, 2023-10-19:
I don't have a clear idea about what `case.wasm` should output. But I'm quite interested in why the different behaviors are came out.
According to the symptom (different file names lead to different behaviors), I think maybe it is caused by the different implementations of `args_get` wasi (or different runtime behaviors when operating the memory `args_get` written to). I think maybe you developers know anything about it. That's why I open this issue.
I'll try to reduce this case to a small one. Hope that can help to find out what caused these different behaviors.

lum1n0us, 2023-10-19:
still not quite sure about what's happening, here are some observations, and a question

Observations:
- use a simple case to test `args_get()` and it works fine. here is the code. Use `/opt/wasi-sdk-20.0/bin/clang-16 -g -o test_args.wasm test_args.c` to generate if you are interested.
``` c
#include <stdio.h>
#include <stdlib.h>

int main(int argc, char**argv)
{
    for (unsigned i = 0; i < argc; i++) {
        printf("argv[%u] = '%s'\n", i, argv[i]);
    }

    return EXIT_SUCCESS;
}

```
- running `iwasm --llvm-jit case.wasm` and `iwasm --llvm-jit $(pwd)/case.wasm` will have two different execution paths.

```
// iwasm --llvm-jit $(pwd)/case.wasm
func[5]
  -> func[04]
  -> func[00]
  -> func[01]
  -> func[30]
    -> func[32]
    -> func[32]
    -> func[32]
  -> func[30]
    -> func[32] 
    -> func[32]
    -> func[32]
  -> func[09]
    -> func[20] 
      -> func[14] 
      -> func[21]  --> raise the exception

// iwasm --llvm-jit case.wasm
`func[5] <_start>`
  -> `func[4]`
  -> `func[0]` <wasi_snapshot_preview1.args_sizes_get>
  -> `func[1]` <wasi_snapshot_preview1.args_get>
  -> `func[30]`
    -> `func[32]`
    -> `func[32]`
    -> `func[31]`
  -> `func[30]`
    -> `func[31]`
  -> `func[9]`
    -> `func[20]`
      -> `func[14]`
      -> `func[21]`
      -> `func[17]`
  -> `func[9]`
    -> `func[20]`
      -> `func[14]`
      -> `func[21]`
      -> `func[17]`
      -> `func[11]`
        -> `func[3]` <wasi_snapshot_preview1.fd_write>
        -> `func[28]`
        -> `func[3]` <wasi_snapshot_preview1.fd_write>
        -> `func[28]`
        -> ...
```

question:
- which combination of compilation option used to generated *iwasm* ?
- because of different execution paths, is there any code in case.wasm which is sensitive with the value of .wasm path?

lum1n0us, 2023-10-19:
> > ___MACOSX/._func.wat_ seems not a regular WAT file. it is a "__MACOSX/._func.wat: AppleDouble encoded Macintosh file".
> 
> I only copy `__stdio_write` function to `func.wat`. You can compare `func.wat` and` func 11` to see what code is added.

any suggestion about how to view/edit func.wat because it is not a text file.

unicornt, 2023-10-19:
> any suggestion about how to view/edit func.wat because it is not a text file.

The .wat file can be opened directly by code editor like VSCode, Xcode. You can also change the suffix name to txt.

unicornt, 2023-10-19:
> still not quite sure about what's happening, here are some observations, and a question
>
> Observations:
>
> * use a simple case to test `args_get()` and it works fine. here is the code. Use [...]

Q1: I believe I didn't use any option except `-DWAMR_BUILD_FAST_INTERP=1`, `-DWAMR_BUILD_FAST_JIT=1`, etc. But I don't remember what the option exactly is.
Q2: Sorry, I have no idea. I think maybe the `insert code` leads to a random access to the memory or change the control flow. 

By the way, how can I get the execution path? It will be helpful for me to analyze it. Thank you!

lum1n0us, 2023-10-20:
> By the way, how can I get the execution path? It will be helpful for me to analyze it.

it is a ongoing PR about [tracing mode]( https://github.com/bytecodealliance/wasm-micro-runtime/pull/2647). Hope you can enjoy it and let me know your opinion.

wenyongh, 2023-11-02:
Hi, I checked the issue again, it should be similar to issue #2602 and #2633: (1) for the hang issue, it should be caused by writing data to STDIN_FILENO, it is allowed in WAMR while prohibited by some other runtimes, and after discussion, we decided not to change the code, see #2602 and #2606. (2) using absolute path and relative path may cause different results of wasi `args_get`, see #2633, it should be reasonable and we don't need to change the code also.

If you apply the patch below, you may get the same result as other runtimes:
```diff
+++ b/core/iwasm/common/wasm_runtime_common.c
@@ -2912,6 +2912,9 @@ wasm_runtime_init_wasi(WASMModuleInstanceCommon *module_inst,
     char *path, resolved_path[PATH_MAX];
     uint32 i;

+    static char *my_argv_buf = "case.wasm";
+    argv[0] = my_argv_buf;
+
     if (!(wasi_ctx = runtime_malloc(sizeof(WASIContext), NULL, error_buf,
                                     error_buf_size))) {
         return false;
diff --git a/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c b/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c
index d5d181f8..df94800b 100644
--- a/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c
+++ b/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c
@@ -730,6 +730,13 @@ fd_table_insert_existing(struct fd_table *ft, __wasi_fd_t in, int out)
 #endif
     }

+    if (in == 0)
+        /* Strip off write bit for stdin */
+        rights_base &= ~(__wasi_rights_t)__WASI_RIGHT_FD_WRITE;
+    else if (in == 1 || in == 2)
+        /* Strip off read bit for stdout and stderr */
+        rights_base &= ~(__wasi_rights_t)__WASI_RIGHT_FD_READ;
+
     error = fd_object_new(type, &fo);
     if (error != 0)
         return false;
```

unicornt, 2023-11-02:
Thank you for the explanation!

closed 2023-11-02 by unicornt