wasm-micro-runtime/wasm-micro-runtime #2646: "Unreachable" exception not thrown correctly on llvm-jit mode

unicornt, 2023-10-16:
### Test Case

[code.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12914070/report4.zip)

### Result

`fast-jit` mode can throw ``unreachable`` exception correctly while `llvm-jit` will throw a `segmentation fault`

[image]

I also ran the same code on other wasm runtimes. They all can report `unreachable` exception correctly.

### Versions and Environment

wamr version : 1.2.3

Operating system: Ubuntu 20.04

Architecture: x86_64

unicornt, 2023-11-01:
> Hi, thanks for reporting the issue! I submitted patch set for PR #2697 to fix, could you please apply it, re-compile wamrc and test again?

I tested it with `llvm-jit` mode and it really works. By the way, issue #2698 I just reported can also be fixed. 
And I'm asking for the cause of this bug again. Could you tell me? Thanks a lot!

unicornt, 2023-11-01:
What is `llvm_entry_block` variable meaning in the code? Is it used to distinguish block like `if0_else` and other type of block?

wenyongh, 2023-11-02:
`llvm_entry_block` is the basic block for the wasm opcode BLOCK, LOOP and the first branch of opcode IF, `if_else` is the basic block of the else branch of opcode IF. Somewhat like:
```wasm
BLOCK/LOOP                        block0_begin/loop0_begin:   (llvm_entry_block)
  ...                      =>        ...
END                               block0_end/loop0_end:       (llvm_end_block)
```
```wasm
IF                        if0_begin:   (llvm_entry_block)
  ...                        ...
ELSE                =>    if0_else:    (llvm_else_block)
  ...                         ...
END                       if0_end:     (llvm_end_block)
```

closed 2023-11-02 by unicornt