wasm-micro-runtime/wasm-micro-runtime #2650: Ignorance of exception "integer divide by zero" in llvm-jit mode

unicornt, 2023-10-17:
### Test Case

[report7.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/12925096/report7.zip)

### Result

If I run this code on `llvm-jit` mode, it will exit normally. But in `fast-jit` mode or `interp` mode, it will threw an `integer divide by zero` exception.
By the way, I tried the same code with other runtimes. They all throw `integer divide by zero` exception.

[image]

### Extra Information

If I change `i32.div_s` (in line 79) into `i32.add`, all runtimes will exit normally. 
Maybe it is a problem of `i32.rotl` instruction because its result is different between `llvm-jit` mode and `fast-jit` mode.

[image]

### Versions and Environment

wamr version : 1.2.3

Operating system: Ubuntu 20.04

Architecture: x86_64

lum1n0us, 2023-10-22:
Because of the `drop` at L80, llvm-jit optimizer treats above Wasm instrctions as a kind of *dead code* and removes them in generation.

unicornt, 2023-10-23:
Thank you for the explanation! Please let me know when and how you fix it.

wenyongh, 2023-11-01:
Hi, I checked the issue again, it seems that it is caused by the i32.rotl opcode but not the i32.div_s opcode. It runs OK after applying patch #2697. For opcode i32.div_s, the AOT compiler adds checks and generates call IR to call aot_set_exception_with_id function, normally it should not be eliminated as dead code.

For example, for the wasm function below:
```wast
(func (export "test") (param i32)
   (i32.const 1234)
   (local.get 0)
   i32.div_s
   drop
)
```
The drop is added after i32.div_s, but after running `iwasm -f test test.aot 0`, the exception `integer divide by zero` was thrown.

unicornt, 2023-11-01:
> Hi, I checked the issue again, it seems that it is caused by the i32.rotl opcode but not the i32.div_s opcode. It runs OK after applying patch #2697. For opcode i32.div_s, the AOT compiler adds checks and generates call IR to call aot_set_exception_with_id function, normally it should not be eliminated as dead code.
> 
> For example, for the wasm function below:
> 
> ```webassembly
> (func (export "test") (param i32)
>    (i32.const 1234)
>    (local.get 0)
>    i32.div_s
>    drop
> )
> ```
> 
> The drop is added after i32.div_s, but after running `iwasm -f test test.aot 0`, the exception `integer divide by zero` was thrown.

Yes, I tried it with patch #2697 and it can throw exception correctly. Thank you for the explanation about `i32.div_s` opcode!

closed 2023-11-01 by unicornt