wasm-micro-runtime/wasm-micro-runtime #2661: Strange "compilation failed" exception in fast-jit mode

unicornt, 2023-10-23:
Hi, I meet a strange  `compilation failed` exception in `fast-jit` mode.

### Test Case
[case.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/13070938/report11.zip)

### Result

I run `case.wasm` in `fast-jit` mode and meet a `compilation failed` exception. And it can run normally in other runtimes(wasmtime, wasmedge, wasmer...).

To trigger this exception again, I tried to put `function 2` into a new wasm file `t.wasm`. I found that `t.wasm` will not throw this exception. 

[image]

### Versions and Environment

wamr version : 1.2.3

Operating system: Ubuntu 20.04

Architecture: x86_64

wenyongh, 2023-10-25:
@unicornt Thanks for reporting the issue, please try PR #2671.

unicornt, 2023-10-25:
> @unicornt Thanks for reporting the issue, please try PR #2671.

Yes, it really works. So this exception is caused by `i32.trunc_f32_s` instruction in `function 4`?

wenyongh, 2023-10-25:
Yes, runtime should check whether the input operand is not nan and in a suitable range, or it  will throw exception:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/interpreter/wasm_interp_classic.c#L592-L599

Here the input operand is a const, and fast-jit generates JIT IR like `cmp i6, const1, const2`, but in the backend, it doesn't handle it:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/fast-jit/cg/x86-64/jit_codegen_x86_64.cpp#L4500

I figured it out by applying optimizations in the frontend as we can pre-know whether to throw exception or not for a const input.

unicornt, 2023-10-26:
Thank you for the explanation!

closed 2023-10-26 by unicornt