wasm-micro-runtime/wasm-micro-runtime #2662: Thread local storage results in data race in AOT mode

Zzzabiyaka, 2023-10-23:
Hi

we're developing some tests for WAMR. 

In process of developing tests that validate thread-local storage behaviour we found that they don't work in AOT mode with TSAN, though WASM mode is fine.

please see log attached:https://gist.github.com/Zzzabiyaka/6f42bc768340a8780f1d05dee4440bf0

Please also see the code that produces the issue: https://gist.github.com/Zzzabiyaka/f58b9a384bf23156b56d265e4d6f8a6a

(the code is quite long itself so I'm sorry in advance). 

to compile wasm from C the following command is used:

```
#!/bin/bash

# This file builds module_test.c using clang and target provided by wasi-sdk
/opt/wasi-sdk/bin/clang -target wasm32-wasi-threads\
    -Werror -g \
    -Wall -pthread -z stack-size=32768 -Wl,--export=__heap_base \
    -Wl,--export=__data_end  -Wl,--shared-memory,--max-memory=1966080 \
    -Wl,--export=wasi_thread_start  -Wl,--export=malloc   -Wl,--export=free \
    -Wl,--export=tls_test \
    --sysroot /home/makslit/t/wasi-libc/sysroot \
    wasi_thread_module_test.c -o wasi_thread_module_test.wasm

```

sysroot is main branch of wasi-libc

to compile aot from wasm

```
./wamrc --opt-level=0 --disable-simd --target=x86_64  --target-abi=gnu --size-level=1 -o wasi_thread_module_test.aot wasi_thread_module_test.wasm 
```
tested against main branch of wamr

Thanks in advance for looking into that

wenyongh, 2023-10-26:
@Zzzabiyaka thanks for reporting the issue, I submitted PR #2673, there is no data race reported again when I tested the aot file, could you try again?

Zzzabiyaka, 2023-10-26:
Sure
Will do today

Zzzabiyaka, 2023-10-26:
No longer reproducible with fix in #2673 . Closing

closed 2023-10-26 by Zzzabiyaka