wasm-micro-runtime/wasm-micro-runtime #2665: Problem building WAMR 1.2.3

lulingar, 2023-10-24:
### Description

When building WAMR-1.2.3 for the x86-64 target, I get `libvmlib.a` without symbols.

### Procedure

Working on a git clone of this repo:
```
$ git checkout WAMR-1.2.3
$ mkdir ${HOME}/wamr
$ cmake -B ${HOME}/wamr \
      -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_FAST_INTERP=1 \
      -DWAMR_BUILD_JIT=0 -DWAMR_BUILD_AOT=1 \
      -DWAMR_BUILD_LIBC_WASI=1 -DWAMR_BUILD_LIBC_BUILTIN=0 \
      -DWAMR_BUILD_LIBC_UVWASI=0 \
      -DWAMR_BUILD_MULTI_MODULE=0 \
      -DWAMR_BUILD_MINI_LOADER=0 \
      -DWAMR_BUILD_SHARED_MEMORY=1 -DWAMR_BUILD_BULK_MEMORY=1 \
      -DWAMR_BUILD_THREAD_MGR=1 -DWAMR_BUILD_LIB_PTHREAD=1 \
      -DWAMR_BUILD_TARGET=X86_64 \
      .
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build Configurations:
     Build as target X86_64
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter enabled
     WAMR AOT enabled
     WAMR Fast JIT disabled
     WAMR LLVM ORC JIT disabled
     Libc builtin disabled
     Libc WASI enabled
     Fast interpreter enabled
     Multiple modules disabled
     Bulk memory feature enabled
     Shared memory enabled
     Thread manager enabled
     Lib pthread enabled
     SIMD enabled
     Reference types disabled
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Stripping static library after build!
-- Stripping shared library after build!
-- Configuring done
-- Generating done
-- Build files have been written to: /home/luis/wamr

$ cmake --build ${HOME}/wamr
Scanning dependencies of target iwasm_static
[  1%] Building C object CMakeFiles/iwasm_static.dir/core/shared/platform/linux/platform_init.c.o
...
[ 48%] Building C object CMakeFiles/iwasm_static.dir/core/iwasm/libraries/thread-mgr/thread_manager.c.o
[ 50%] Linking C static library libvmlib.a
[ 50%] Built target iwasm_static
Scanning dependencies of target iwasm_shared
[ 51%] Building C object CMakeFiles/iwasm_shared.dir/core/shared/platform/linux/platform_init.c.o
...
[100%] Linking C shared library libiwasm.so
[100%] Built target iwasm_shared
```

Then I check what symbols made it into `libvmlib.a`:
```
$ nm ~/ram/wamr/libvmlib.a
nm: wasm_application.c.o: no symbols
nm: wasm_c_api.c.o: no symbols
nm: wasm_exec_env.c.o: no symbols
nm: wasm_memory.c.o: no symbols
nm: wasm_native.c.o: no symbols
...
```
No symbols!

I have also used clang-14 for the build. Same outcome. It was highly unlikely that would made a difference, but I tried it nonetheless.

lulingar, 2023-10-24:
We realized that we absolutely need to add the `-DCMAKE_STRIP=0` option.
While that works, it's a workaround, it seems to point to a mistake in the build.

wenyongh, 2023-10-25:
@lulingar thanks for reporting the issue, a patch has been submitted to fix it: https://github.com/bytecodealliance/wasm-micro-runtime/pull/2669

lulingar, 2023-10-25:
Thank you @wenyongh! I'm closing this issue.

closed 2023-10-25 by lulingar